cedricp/ddt4all #798: Downgrade from 7.0.24.161 to 3.3.16.961 failed

AndreiBBF, 2023-03-24:
Hello! I ve made a mistake. I ve downgraded my rlink2 from 7.0. to 3.3 and now my reverse camera is not working at all, parking sensors are without sound and not appear on the display, temperature is shown in F beside C, can t turn off navigation from button. I ve tried all versions of firmware but always got this message: " It is not possible to install a former software version, current software version is kept “. Can anyone give me an advice?
P.s. Emergency  download is not working.
Thanks!
[image]
[image]

rttr67, 2023-03-25:
enter your email address

BBFul, 2023-03-25:
work.acces@yahoo.com
 Thanks!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

rttr67, 2025-05-08:
I can help.
The device needs to be replaced

rttr67, 2025-05-08:
I can help.
The device needs to be replaced
zaliavos.ud@gmail.com

AlMeu, 2025-05-08:
> Hello, Allow me to bump this thread. I'm facing the same error when trying to downgrade from v8 to v3, after an incorrect upgrade to v8 with an incompatible bootloader. Could you please share how you managed to solve your issue? Thank you very much!

Yes, only replace unit you can do in your case.
You have to get a new unit or old one unlocked.

bajdenis995-cyber, 2026-09-30:
Добрый день 
Я столкнулся с  проблемой после обновления с v3 до v9 Пропал звук радио,звук парктроников и задня камера 
пожайлусто подскажите как починить Renaut Scenic 4

rttr67, 2026-09-30:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

только замена радиомодуля