cedricp/ddt4all #799: Radio not turning off automatically in Clio IV after MediaNav Evolution 2 replacement

JakubUrbanowski, 2023-03-25:
Hello everyone, I have a Clio IV with a wireless key. I have replaced my MediaNav Evolution with a MediaNav Evolution 2. Previously, the radio would turn off when I turned off the engine and opened the door. Now, the radio stays on and I have to turn it off manually.

Does anyone have an idea which configuration could be causing this? I have only configured the ECU ULC_gen4_v.1.4 for this. I do have the "digital radio antenna phantom supply" set to active instead of passive. Could this be the problem?

Thank you in advance for your help.

AndyIreland, 2023-03-25:
have you tried MICOM MANAGER on your evolution - it my have door setting in that. In my clio mk4 it has the evolution (9.1.3 - what is your software version on yours?)  ( mine with DAB but no AA & CP)  and I can get into Micom manager with that  . I think you can use micom manager with the later evolution with AA & CP.

JakubUrbanowski, 2023-03-25:
I’m using 

> have you tried MICOM MANAGER on your evolution - it my have door setting in that. In my clio mk4 it has the evolution (9.1.3 - what is your software version on yours?) ( mine with DAB but no AA & [...]

I have an MN3 V1.0.15.3. What settings would this be within the MiCom? And would I also be able to set up parking sensors here?

KarelSvo, 2023-03-25:
DataWrite.WAKE UP
Wake Up from SYSTEM ON

JakubUrbanowski, 2023-03-25:
Thank you! This did the trick. And how can I enable parking sensors within MiCom? 

This is my MiCom screen 
[image]

AndyIreland, 2023-03-25:
here is my 9.1.3 micom manager (its earlier than your medianav) and I think parking sensors on mine is SND or HMI i cant remember .. (and our clio mk4 hasnt got UPA sensors either) 
[image]

AndyIreland, 2023-03-25:
your one surely would be the bottom row 

UPA SND, UPA SPK and UPA HMI for Universal parking sensors

JakubUrbanowski, 2023-03-26:
For me it was only the button UPAHMI. 

Thank you for helping!

closed 2023-03-26 by JakubUrbanowski