cedricp/ddt4all #801: Center console ambient light not working after screen swap

Lauddt94, 2023-03-29:
Hi everone, 

I have an issue with my megane IV after changing my 7’’ display to the 8.7’’ version. 

Everything seemed fine but after a while I discovered that the my sense ambient led strips on the center console weren’t working anymore. All the other led strips in the doors are still functioning. My best guess would be that something has to change in the settings because a hardware malfunction for two strips at the same time seems unlikely.

I tried all my sense / ambient light settings I could find using ddt4all.

Is anyone familiar with this issue after changing a display? Or does someone know how I can check or solve this problem? Thanks a lot.

AndyIreland, 2023-03-29:
what is it a rlink2 on it? what radio /nav ECU shows up when you do a search with the DDT4ALL magnifying glass?

Lauddt94, 2023-03-30:
Yes it is a Rlink 2. When I look with the magnifying glass is say: Found ECU [audio] (no relevant ECU file found) :DIAGVERSION [0D] SUPPLIER [CAP] SOFT [90] VERSION [34611]

An additional issue occured, the climate control is functions normal for a couple of seconds. After a couple of seconds it only blows hot air unless you turn it to LO. 

Maybe is there some list with all default settings for a Megane IV?

Lauddt94, 2023-03-30:
To be honest I think I messed up some settings with the audio and HVAC configuration. At some point there was no screen, no radio and no sound comming out of the speakers. Most things I figured out myself but the Hot air form the blower and the center ambient lights are still a problem. Is there anyone who can help with solving the problems mentioned above, I have the screens.txt  available. Thanks

shrlnm, 2023-03-30:
@Lauddt94 I may try to help but I need the dump from RadNav given by pyren.

Lauddt94, 2023-03-30:
@shrlnm Is this the screens.txt document?

Lauddt94, 2023-03-30:
@shrlnm thanks for helping! Attached is the screens.txt and I will install pyren and look for the dump file. 
[Uploading screens.txt…]()

shrlnm, 2023-03-30:
@Lauddt94 have you changed r-link2 too? Is the VIN number of your car ends with 6469?

shrlnm, 2023-03-30:
@Lauddt94 I suppose I can't help without current pyren dump. Too many unknowns regarding radnav

Regarding the climate, I suppose you have to change the value CL_WithControlPanel  in ClimBoxGen2_v22_20160218T104803 from 7 to 8

Lauddt94, 2023-03-30:
@shrlnm 
[image]
I am trying to figure out how to make a dump using pyren. See the picture attached for the software version.

Lauddt94, 2023-03-30:
@shrlnm 
[image]

I am trying to figure out how to make a dump using pyren. See the picture attached for the software version. 

Software release: 9.0.34.611
Boot: 5615

shrlnm, 2023-03-30:
Hmm. 5615 is for 7.0
In order to be sure that all your changes made via CAN are applied, you will first need to downgrade to the version of LGU that you were supplied with 5615
For making dump in pyren just enable appropriate parameter in a launcher. Also in your case it would be need to make a dump also in mod_ddt as it contains more data

Lauddt94, 2023-03-30:
@shrlnm when I bought the car it had V2 or V3 software if I can remember. I updated myself from 7 to 9 and that is already some time ago. The 7” was working correct on this software and boot and so did the ambient light. Do you think it is necessary to downgrade?

Lauddt94, 2023-04-14:
Last week problem solved, item can be closed

closed 2023-04-14 by Lauddt94

Silvstoian, 2024-04-16:
> Last week problem solved, item can be closed

Hi, how did you fix it ?

ragippoker, 2025-01-09:
I have the same problem. How did you solve it? ragippoker@gmail.com