cedricp/ddt4all #802: I can't figure out how to configure WAKE on my MediaNav3 correctly

JakubUrbanowski, 2023-03-30:
Dear all,

A friend of mine owns a Renault Clio with a wireless key. When he turns off the engine, the radio also turns off immediately. Previously, this only happened when he opened the car door. I have set WAKE to ON in the MiCom. Does anyone happen to know if there is still something incorrectly configured within the MiCom?

[image]

Thank you in advance!

Jakub

AlMeu, 2023-03-30:
Hi Jakub,
Change the status of **Wake** button pressing shortly and acknowledge with Hard Reset

JakubUrbanowski, 2023-03-31:
Hello @AlMeu ,

The MiCom screen from the picture is the current configuration. But that is the correct configuration right? Because I tried already to reset hard after setting WAKE ON. But the radio is still turning off when stopping the engine instead of when opening the door

AlMeu, 2023-03-31:
@JakubUrbanowski change the status of Wake button pressing shortly I said.
If it was ON, switch it in OFF acknowledge it with Hard Reset and check again.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

an0nimo, 2024-02-09:
> Dear all,
>
> A friend of mine owns a Renault Clio with a wireless key. When he turns off the engine, the radio also turns off immediately. Previously, this only happened when he opened the car door. I [...]

Hi @JakubUrbanowski.
I don't know if you have already found a solution but I had this same situation after a 40,000 km review of the car and the change in the BIC/A2/A3 module disabled the appearance of the digital clock on the Media Nav screen when opening the doors as well as turn off Media Nav when leaving the car and close the doors using the key control.
In addition to the "WAKE ON" button as in the image you posted, you will need to access the MEDIA NAV settings using the DDT4ALL + ELM327 scanner or the VGATE OBD2.
As the MEDIA NAV is on a different CAN than the car modules, it will also be necessary to make a change in the connection between 02 pins of the CAR OBD SOCKET and the ELM or VGATE scanner.
There are people who changed the wires inside the ELM and others, like me, bought an OBD splitter to change the position of the wires more easily.
In the link below is all the support I got here regarding the use of DDT4ALL + ODB splitter + ELM327 scanner to reprogram my MEDIA NAV:
https://github.com/cedricp/ddt4all/discussions/970
In the link below are the changes I made using DDT4ALL in the BIC/A2/A3 module to make the digital clock appear again on the MEDIA NAV screen as soon as I open the doors and also turn it off as soon as I close the doors using the control on the vehicle key. car:
https://github.com/cedricp/ddt4all/discussions/1026

I hope this helps you.