cedricp/ddt4all #803: no access to ecu of Dacia Sandero stepway 2021 with OBDLink SX

labdomenico, 2023-03-31:
Good morning,
my OBDLink SX passes all ddt4all tests (52/52). It connects regularly to the pc but it does not allow me to write and read data in the ecu.
Please tell me if I'm doing something wrong in the linking process.
1) Specially formatted pc, win 11, ddt4all installed, OBDLink driver installed, green OBDLink led.
2) I turn on the pc, connect the usb to the pc, connect OBDLink to the car
3) I start the engine, select the right com port and connect ddt4all
4) if I look for the ecu it doesn't give me anything;
5) if I select logan/sandero 3 and I look for the ecu it doesn't give me anything.

It's the third adapter I've bought because I thought the other 2 weren't suitable for my car.

Is there a way to inhibit the battery saver and be able to work with ddt4all with the engine off (and instrument panel on) without everything turning off? for example, I know that for megane 4 you have to put a gear and keep the start/stop button pressed for 5 seconds. My sandero has traditional iron keys.
Thank you

KarelSvo, 2023-03-31:
> It connects regularly to the pc but it does not allow me to write and **read data in the ecu**.

[Click here](http://logan3wiki.wikidot.com/wiki:ots-rechemari#toc27)

labdomenico, 2023-04-04:
thank you all.
the question is: if i buy the renault-key for the gateway, can i work with obdlink sx? (https://asos.renault.com/#/presentation).
If obdlink can work in pass-thru then it works.
if you don't understand what I mean you can get an idea here https://www.youtube.com/watch?v=MaLw8DO3gTo

KarelSvo, 2023-04-04:
obdlink is not a passthru interface

labdomenico, 2023-04-05:
[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif