cedricp/ddt4all #805: Maybe it is a time to create new release?

igogo2, 2023-04-03:
Latest v2.0 doesn't work correctly.

AndyIreland, 2023-04-03:
try visiting here for updates?
https://github.com/Furtif/ddt4all

igogo2, 2023-04-03:
Try to count how many commits is there https://github.com/cedricp/ddt4all/commits/master since Nov 10, 2020.

igogo2, 2023-04-05:
@cedricp

igogo2, 2023-04-09:
How i can run the application? What i do wrong?
[image]

AndyIreland, 2023-04-09:
if you are trying to run the DDT4ALL for windows Application and you want one with a built in database and desktop icon use this one: 
https://www.drive2.ru/b/498093336985338243/

igogo2, 2023-04-09:
Thanks. But i want to run the newest version on Python 3. Not from 2019 year.

AndyIreland, 2023-04-09:
> Thanks. But i want to run the newest version on Python 3. Not from 2019 year.

why? :) 
 if it does what you want it to do and finds the right ECU's parameters that you want to change?
 
 There is another workaround if you want to have up to date and i think this will work , you can download from Drive 2 so that you have the database and the desktop icon then go to furtive updates on github and download all the files then go to your program files/ddt4all folder and paste the up to date files over-writing the files ... I think that will work

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif