cedricp/ddt4all #806: Renault Clio mk4 check brake light message

z0523, 2023-04-06:
Hi everybody!
I have a clio mk4 2013 (1.5dci).  After a usual braking i got the “check brake lights” error message and the wrench icon on the dashboard. Ofc i checked brake lights, cleaned the contants between the lamps and bulbs, and replaced the bulbs (no LED). Everything works well. I used delphi ds150 to read the error codes. I have an error “dtc9259 - brake light circuit”. After deleting the codes, the message and the wrench icon reappears after pressing the brake pedal a few (varies between 3-30) times. I replaced the brake pedal switch, the front brake discs and pads, cleaned the sensors in the front and the back, replaced the master cylinder, replaced the right rear lamp, replaced some of the fuses connected to brakes. The initialization of the ecu was also done. And the message always come back. I removed the bulbs and turned on the lights to check the voltage in the brake lamps. It was about 4v on the left side and 28v (!!!) on the right side without pressing the brake pedal. After pressing the pedal it was about 11.5v on both sides.
I think also the start/stop doesn’t work properly. Sometimes its ok, but sometimes after 30-40 kms it still says that its unavailable while ac is off and the external temperature is 15C. Maybe there is a connection between the problems. 
Any advice on how to solve this problem?

AndyIreland, 2023-04-06:
I would try clearing the DTC errors with PyREN and an obd2 adaptor - see if they come back then. 
See if it says 'memorized' or 'active' for the DTC faults

z0523, 2023-04-07:
> I would try clearing the DTC errors with PyREN and an obd2 adaptor - see if they come back then. See if it says 'memorized' or 'active' for the DTC faults

I tried PyREN. It's an active fault.

[image]

[image]

It was deleted, and still comes back. It's new information also for me, that the fault appears only when the engine is running.

The DF121 code is also new. Can it be caused by the fault of the knock sensor or the vacuum pipe sensor or the brake fluid pressure sensor? 
I forgot to mention that an Autel diagnostic showed some mastervac problem (i dont remember what was that exactly, but the fault code waws the same DTC925909)

> > I used delphi ds150 to read the error codes.
> 
> Try to read DTCs by the pyren3. Get documentation by t.me/pyren_bot , find all DTC codes there and procedures how to solve them.

I got the mentioned documentation. My error code is not listed.

AndyIreland, 2023-04-07:
comp. = computer/ECU (electronic computer unit) - so it says its faulty. that would be your UPC/EMM ECU - maybe it has a hardware fault?

z0523, 2023-04-08:
Ok, now I have some stupid quesitions.
What are the chanches of that this is really hardware falut?
Can an update solve the problem (is it possible anyway?)
Can hardware faults be detected by measuring, or any other way?
Is it possible to repair  those units?

shrlnm, 2023-04-08:
@z0523  I suggest to check wiring. If there is no info about DTC925909, then find there info about any DTC "STOP LIGHTS CIRCUITS" and use procedure for wiring check from it. If the wiring is good then the problem is in ECU and I not sure that software upgrade may solve it. Then you have two choices. 1) change the ECU 2) try to disable (inhibit) checking of stop lights circuit. Usually it may be done by ddt

AndyIreland, 2023-04-08:
> Ok, now I have some stupid quesitions. What are the chanches of that this is really hardware falut? Can an update solve the problem (is it possible anyway?) Can hardware faults be detected by measuring, or any other way? Is it possible to repair those units?
> 
> Can it be a software fault in the ECU? If yes, and the ECU will be replaced, will be the fault copied during the cloning process?

yes I would say it would be hardware rather than software (wiring of car or ECU itself)

I would check other parameters as well like making sure its not set to 'stop with PWM' or stop by LED  and all settings similar in the emm ecu

z0523, 2023-04-09:
I checked the mentioned settings. The nbx_tail_1_with_LED_cf and nbx_tail_2_with_LED_cf are set to 1. The nbx_tail_3_with_LED_cf set to 0. None of them can be changed by ddt. I think it would be the best to disable chenking of brake lights. How can i do that?

shrlnm, 2023-04-09:
> I checked the mentioned settings

Which xml do you use? Or post your vin

z0523, 2023-04-09:
I don’t know the answer for xml. 
vin: vf17rjl0h48879114

AndyIreland, 2023-04-09:
> I don’t know the answer for xml. How can i check that anyway? vin: vf17rjl0h48879114

find your ecu's (xml) by using the magnifying glass in DDT4ALL

shrlnm, 2023-04-09:
> vin: vf17rjl0h48879114

xml T4_EMM_DDT2000_SW8_V2.1.xml

Most relevant parameters are
##     Nxx_BRK_PDL_Stop_Lamp_FIh_CF 1 = 1 #[DTC_STOP_LAMP_OC_Ext : ]
##     Nxx_BRK_PDL_Stop_Lamp_FIh_CF 2 = 1 #[DTC_STOP_LAMP_SCG_Ext : ]
##     Nxx_DTC_FIh_CF 15 = 1 #[DTC_Stop_Lamp_OverLoad_DTC_Ext : ]
but I do not know exactly

AndyIreland, 2023-04-09:
heres the ecu I would look on mine , but my SW is SW15_1 ... you may have an earlier one and not listed I dont know. But if anything I would experiment with these settings if they show up on your SW version ECU @z0523 

[image]

z0523, 2023-04-10:
> > vin: vf17rjl0h48879114
> 
> xml T4_EMM_DDT2000_SW8_V2.1.xml
> 
> Most relevant parameters are Nxx_BRK_PDL_Stop_Lamp_FIh_CF 1 = 1 #[DTC_STOP_LAMP_OC_Ext : ] Nxx_BRK_PDL_Stop_Lamp_FIh_CF 2 = 1 #[DTC_STOP_LAMP_SCG_Ext : ] Nxx_DTC_FIh_CF 15 = 1 #[DTC_Stop_Lamp_OverLoad_DTC_Ext : ] but I do not know exactly

I set those settings to . The fault still returns.

> heres the ecu I would look on mine , but my SW is SW15_1 ... you may have an earlier one and not listed I dont know. But if anything I would experiment with these settings if they show up on your SW version ECU @z0523
> 
> [image]

I tried to change also these settings, but my SW is older than yours and I have much less options.
[image]

AndyIreland, 2023-04-10:
@z0523  Normally I don't suggest manually trying out a ECU that is not found automatically but what if you tried selecting T4 VS BCM DDT2000 SW15_1 and  stop pedal configuration and look at what it reads (if its all in red colour then obviously you wont be able to change it) - **change parameters at your own risk** just in case you brick the ECU
other than that if your SW is older and does not have the parameters listed there at all to change then I dont know what else you can try

z0523, 2023-04-11:
I selected SW15_1. The results:

[image]

I didn't dare to change values where "no data" is showed. To say the truth i don't know if it's possible. I also changed the last 3 values to false. It didn't work.

AndyIreland, 2023-04-11:
no, never change the ones in red in the 'read' (even if its even possible) 

so after you changed the last 3 parameters did you do a refresh and see if the settings had kept? - then afterwards clear the DTC's ?

AndyIreland, 2023-04-11:
OK you say first you cleared DTC then made changes .... did you then clear DTC again after making the changes?

z0523, 2023-04-11:
Only after the fault came back after pushing the brake pedal (this is the only way to test that the procedure was successful)

AndyIreland, 2023-04-11:
> Only after the fault came back after pushing the brake pedal (this is the only way to test that the procedure was successful)

I have a feeling if you have checked all the wiring I reckon you might have to be changing the T4 BCM ... but maybe see if you can change it to a later SW15_1 if thats possible (although it could be tied to the car it came off with the VIN number, I am not sure?)  - but at least you will have more settings available

z0523, 2023-04-11:
I can not check further the wiring, but the visible parts seemed to be fine. I think i don’t have the equipment and the sowtware either to do the update myself. I only have a single ELM327 adaptor and I think it’s not possible to do with that. The whole process is so stressful and time consuming, i’ve decided it is best to take it to a garage. Although i am not sure they can fix it either. Anyway, i will keep you posted.

AndyIreland, 2023-04-11:
good luck hope you get it sorted all out OK

z0523, 2023-04-11:
I’m so grateful for your help! Thank you, thank you, thank you. I will definitely share the information if the problem will be solved. 
[image]

z0523, 2023-04-14:
As I promised, I'm back :)
It was told by the garage, that the module in the pic (I think it's the UCH) needs to be replaced.

[image]

Now I have to order the module. They told me that only number marked with yellow have to match. Do I think well, that the number marked with red also has to be the same? Do they have to clone the old module when it will be replaced?
I found a few with matching numbers, but some of them came from petrol cars or diesel ones but with different engines. Does it matter or i just have to check the part numbers?

AndyIreland, 2023-04-29:
ah right yes that would cause it then LOL 

glad you got got it sorted out though

closed 2023-04-29 by z0523