cedricp/ddt4all #807: Renault scenic 3 R link EOL settings add

zlocko795, 2023-04-07:
After accidentally resetting r link, I keep getting this message, is there any help?

shrlnm, 2023-04-07:
@zlocko795 https://www.drive2.ru/b/643019689764615344/

AndyIreland, 2023-04-07:
just 7 hours ago earlier this user said he fixed his rlink1 by just randomly putting in user and password and it worked:
https://github.com/cedricp/ddt4all/issues/783#issuecomment-1500131322

zlocko795, 2023-04-07:
I can't find the ecu:rad nav, please help

shrlnm, 2023-04-07:
@zlocko795 Not sure that understand you well.  You wrote about R-Link1 right?  RadNav it is R-Link2
What is your vin? What is the problem now?

zlocko795, 2023-04-07:
my Vin number is vf1jzm50652520404
I looked at the picture on the website that you sent me, so I thought it was the same

AndyIreland, 2023-04-07:
hasnt Renault Scenic 3 2015 got rlink1 in it?

zlocko795, 2023-04-07:
Yes yes it has r link 1.
Sorry if I'm writing wrong, I know intermediate English

AndyIreland, 2023-04-07:
> Yes yes it has r link 1. Sorry if I'm writing wrong, I know intermediate English

ah right thats OK . understandably things can get confusing in different languages . 

If you want to try using this as a possible fix you can try - it might work, it might not. It got my rlink1 working when i had an EOL message once on my car:
try SD card way first. Format a 8gb SD card (not your map card!!) to fat32 and put tomtom.000 on the root.
If no work try the USB way format a 8GB Usb stick to Fat32 and put tomtom.000 on the root

keep car/engine running while its loading the file , it could take 20mins to load - after its finished loading manually reboot the rlink by pressing home button 5 times.

https://we.tl/t-PKn3pjjISn

zlocko795, 2023-04-08:
I tried that but it won't start loading, I'll try again with the file you sent me

zlocko795, 2023-04-08:
I did as @shrlnm  told me, but now when I turn on the r link radio it only works for 10 seconds, what is the problem

zlocko795, 2023-04-08:
I solved the problem with the radio by re-inserting the USB TOMTOM.000, and the radio did not turn off after 10 seconds.  Thank you for your great help

AndyIreland, 2023-04-08:
have you re-opened it because its still not working? @zlocko795

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif