cedricp/ddt4all #808: DDT4ALL with Engine on + RED/GREEN ECUs

Trickykid1978, 2023-04-10:
Good morning everyone ... 

I have two questions based on my first contact with DDT4ALL connected mode : 

1) is it ok to connect DDT4ALL with engine on ? (not just ignition ON) . Apparently, i want to by pass the auto/forced Ignition OFF  of my Clio IV 

2) RED and GREEN ECUs :  if i use the magnifying glass and CAN mode to scan my ECUs , i am getting some green and red ones (reds reported as not a perfect match) while other ECUs do not even come up . But when i choose my car model from the drop down menu  without using the magnifying glass, then i apparently get full access to more ECUs without green/red flags . If i save a red ECU in my ECUs folder of DDT4ALL  , will it then appear as green if i rescan my car ? Finally and most importantly,  is it safe to change values on red ecus ? (at least the red ones reported as Not perfect match) .

Trickykid1978, 2023-04-10:
> theory or practice? Why don't you copy the ecu scan list from the ddt4all log window here?

Hi, well the reason is because i left the laptop in my car :)  ...in any case, my question is not case-specific : which is the safest method to use ddt4all ? through the model list of the drop down menu or through the magnifying glass ? in the first case, you won't get any green and red ecus and you will get a full list of ecus (either with data or not) ... when i use the magnifying glass, i am getting 5 ecus in total (2 greens-3 reds or sth) . For example, the magnifying glass fails to find a specific ecu (injection) ...if i load this specific ecu through the model selection menu, i am getting correct data (which means that ecu is alive and ddt4all is manually connected to it) ...

AndyIreland, 2023-04-10:
there is a thing called 'park mode' which you could try so that the car dont go to sleep @Trickykid1978  - Basically you put your keycard into the keycard slot on the dashboard then put the gearstick into 1st gear (if manual gearbox car) or 'D' (drive if its an auto) and (without pressing any pedals) long press the 'START' button on the dash and then 'Please Remove Keycard' should show up in the speedometer cluster - now (if it works as it should and you leave the keycard in the slot) then by rights the car (should) stay on and not time out (IE not go to sleep )

Trickykid1978, 2023-04-10:
> there is a thing called 'park mode' which you could try so that the car dont go to sleep @Trickykid1978 - Basically you put your keycard into the keycard slot on the dashboard then put the gearstick [...]

Hi Andy, 

I did try that but it didn't work ... maybe i was pressing the brake pedal, not sure so i will give it another try tomorrow ...in any case, the start stop deactivation i am planning to do tomorrow won't take more than 60 seconds to complete ... after posting this, i did operate ddt4all with engine on with no issues (at least with reading ecus) . One strange thing i noticed was in the AT ECU of the dual clutch gearbox where ECO Mode was reported ...i then pressed the ECO button in the car and rescanned ecu and then, it was reported as in SNOW mode (wtf ?) ...

AndyIreland, 2023-04-10:
> Hi, well the reason is because i left the laptop in my car :) ...in any case, my question is not case-specific : which is the safest method to use ddt4all ? through the model list of the drop down [...]

always dodgy to choose ECU's manually from the drop down list (although I have had to do it in the past) take for example when I wanted to enable Android Auto on my car well I had to manually select the MFD to enable it, and  the same with the DCM TCU, would not show when I used the magnifying glass so I had to choose it manually. 

One other thing you can try is, before you search with the magnifying glass, is to change the CAN timeout to 600ms and see if that finds more ECU's

Trickykid1978, 2023-04-10:
> manually select the MFD to enable it, and the same with the DCM TCU, would not show when I used the magnifying glass so I had to choose it manually.
> 
> One other thing you can try is, before you search with the magnifying glass, is to change the CAN timeout to 600ms and see if that finds more ECU's

That's a good idea, thanks ....

AndyIreland, 2023-04-10:
> Hi Andy,
>
> I did try that but it didn't work ... maybe i was pressing the brake pedal, not sure so i will give it another try tomorrow ...in any case, the start stop deactivation i am planning to do [...]

oh right - have never heard of 'snow mode' that sounds something like you would have on the ASR ESC system rather than the eco one or AT ECU

shrlnm, 2023-04-11:
> which is the safest method to use ddt4all ?

Hi
The safest method is the next one
1) use search procedure "magnifying glass" for finding all ECUs with full match IDs (GREEN modules)
2) check all RED modules by REPROG DB.  Ask the pyren_bot your VIN and it returns you names of  ecus (in format of ddt database) which were used by manufacturer engineers for preparation initial configuration for your ecus.
3) choose the closest names from your ddt db to names from reprog db. 

And, of course, always keep a log of any of your actions

AndyIreland, 2023-04-11:
>which is the safest method to use ddt4all ?

well one of the safest seeing as DDT4ALL does not have a backup facility is to install PyREN and backup (dump /.txt) of all your ecu's before even changing parameters through DDT4ALL (you can also use PyREN in DDT4ALL mode / select DDT4ALL screens when you are in the ECU with PyREN)

Trickykid1978, 2023-04-11:
Hello everybody, here below are the ecus identified by the magnifying glass (after setting CAN timeout to 800ms) . Is there a straightforward and step-by-step  way to turn the RED Cluster and T4_VS_BCM ecus in GREEN ecus ? (these are the two ecus i will mess around and both versions exist in my database) . 
[image]
[image]

KarelSvo, 2023-04-11:
Yes, of course i can make everything green, but you didn't copy the result rows.
This is a developer database with a lead time of about 1 year before production. All actually produced software versions are never stored there. Sometimes there are also spelling errors in the automatic identification data.

KarelSvo, 2023-04-12:
Just copy and paste the green and red lines from your photo to your answer.

Trickykid1978, 2023-04-12:
> Just copy and paste the green and red lines from your photo to your answer.

ok , i will do so tomorrow morning ... many thanks for your assistance !

Trickykid1978, 2023-04-14:
> Just copy and paste the green and red lines from your photo to your answer.

@KarelSvo 

Here are the results : 
Scanning CAN
Found ECU [BCM/UCH] (not perfect match) :T4_VS_BCM_DDT2000_SW16_5 DIAGVERSION [5] SUPPLIER [001] SOFT [1419] VERSION [1030 instead 1310]
Identified ECU [UPC-EMM]@27 : EMM_EDISON_DDT2000_SW15_1_V1.0_20160531T142349.json DIAGVERSION [8]SUPPLIER [414] SOFT [C151] VERSION [7000] {2572}
Found ECU [HVAC] (no relevant ECU file found) :DIAGVERSION [09] SUPPLIER [B517] SOFT [SW06.B.06-B019-0] VERSION [C988110]
Identified ECU [Airbag]@2C : RSAT4_ACU_eng_v15_20150511T131328.json DIAGVERSION [4]SUPPLIER [AMR] SOFT [0470] VERSION [0470] {6121}
Found ECU [Cluster/TDB] (not perfect match) :Cluster X98_X87_X82_xHA_X98ph2_X87ph2_xHAph2_v8.02 DIAGVERSION [28] SUPPLIER [ANC] SOFT [0062] VERSION [2005 instead 0908]
Found ECU [Power steering] (not perfect match) :X07_EPS_V3.0 DIAGVERSION [4] SUPPLIER [373] SOFT [D201] VERSION [1200 instead 200B]
Found ECU [EPS] (not perfect match) :X07_EPS_V3.0 DIAGVERSION [4] SUPPLIER [373] SOFT [D201] VERSION [1200 instead 200B]
Identified ECU [HFM]@C0 : T4_VS_HFM_DDT2000_SW14.45_20170808T141805.json DIAGVERSION [5]SUPPLIER [001] SOFT [1420] VERSION [1435] {6930}
Identified ECU [AT]@6E : DC4_X98_X95_X87_X61_L38_EU6_vdiag0C_V1.4_20160309T101106.json DIAGVERSION [12]SUPPLIER [487] SOFT [3000] VERSION [3600] {1608}

Trickykid1978, 2023-04-14:
I also tried to clear some DTCs from ECU [HFM]@c0 : T4_VS_HFM_DDT2000_SW14.45_20170808T141805.json DIAGVERSION [5]SUPPLIER [001] SOFT [1420] VERSION [1435] {6930}  but it kept on failing : 

DTCs : 
ECU trouble codes
DTC #0
DTCStatus.pendingDTC : [00] No
DTCDeviceIdentifier : [a003] RF
DTCStatus.testFailedSinceLastClear : [00] No
DTCDeviceAndFailureTypeOBD : 40963 [0xa003]
DTCFailureType.ManufacturerOrSupplier : 8 [0x8]
DTCFailureType.Category : [03] FM (Frequency Modulated) / PWM (Pulse Width Modulated) Failures
DTCFailureType : [38] signal frequency incorrect
DTCStatus.testNotCompletedSinceLastClear : [01] Yes
DTCStatus.testNotCompletedThisMonitoringCycle : [00] No
DTCStatusAvailabilityMask : 185 [0xb9]
DTCStatus.warningIndicatorRequested : [00] No
StatusOfDTC : 16 [0x10]
DTCStatus.testFailedThisMonitoringCycle : [00] No
DTCStatus.testFailed : [00] No
DTCStatus.confirmedDTC : [00] No
DTC #1
DTCStatus.pendingDTC : [00] No
DTCDeviceIdentifier : [a010] PWR_ECU
DTCStatus.testFailedSinceLastClear : [00] No
DTCDeviceAndFailureTypeOBD : 40976 [0xa010]
DTCFailureType.ManufacturerOrSupplier : 4 [0x4]
DTCFailureType.Category : [01] General Electrical Failure
DTCFailureType : [14] circuit short to ground or open
DTCStatus.testNotCompletedSinceLastClear : [01] Yes
DTCStatus.testNotCompletedThisMonitoringCycle : [00] No
DTCStatusAvailabilityMask : 16 [0x10]
DTCStatus.warningIndicatorRequested : [00] No
StatusOfDTC : 16 [0x10]
DTCStatus.testFailedThisMonitoringCycle : [00] No
DTCStatus.testFailed : [00] No
DTCStatus.confirmedDTC : [00] No

and results : 

Clearing DTC information
Sending ELM request :14FFFFFF
ELM response : 54 
Clear DTC successfully done
Sending ELM request :1902FF
Bad ELM response : WRONG RESPONSE : Unknown(5902B9A0033810A0101410A03A1410A03B1410A03C1410A03D1410A03E1410A03F1410A0401410A0411410A0429610A0500110A0510410A0514910A0514A10A0516210A0517710A0518510A0518710A0525210A0525510A0526710A0528710A0528810A0529710A0529810A0531210A0531C10A0541210A0)
ELM response : WRONG RESPONSE : Unknown(5902B9A0033810A0101410A03A1410A03B1410A03C1410A03D1410A03E1410A03F1410A0401410A0411410A0429610A0500110A0510410A0514910A0514A10A0516210A0517710A0518510A0518710A0525210A0525510A0526710A0528710A0528810A0529710A0529810A0531210A0531C10A0541210A0)

Can anyone guess why it fails to clear DTC from this ecu ? 

Not sure if it is relevant, but when i benchmark my ELM327 USB , it fails in all five STN11xx TESTS (it passes 47 of 52 tests in total)

shrlnm, 2023-04-14:
> Can anyone guess why it fails to clear DTC from this ecu ?

> DTCStatus.testNotCompletedSinceLastClear : [01] Yes

shrlnm, 2023-04-14:
@Trickykid1978 You may read how it works, for example there 
https://piembsystech.com/diagnostic-fault-code-status-of-dtc-in-a-vehicle/

Trickykid1978, 2023-04-14:
> in few words, you can't confirm the DTC for Stop Lamp or flashers while you don't press the stop pedal or switch on a flasher. It is not real trouble. ECU did not have a chance to do a test.

Now i get it , many thanks . Both errors (40963 and 40976) refer to handsfree so i also don't think are real  DTCs ...

AndyIreland, 2023-04-14:
> in few words, you can't confirm the DTC for Stop Lamp or flashers while you don't press the break pedal or switch on a flasher. It is not real trouble. ECU did not have a chance to do a test.

thats a good tip!

AndyIreland, 2023-04-14:
> I also tried to clear some DTCs from ECU [HFM]@c0 : T4_VS_HFM_DDT2000_SW14.45_20170808T141805.json DIAGVERSION [5]SUPPLIER [001] SOFT [1420] VERSION [1435] {6930} but it kept on failing :
>
> DTCs : [...]

HFM= Hands Free Motoring (like module for remote key doorlock entry and wireless/keyless start)
Try clearing the DTC using PyREN.
Also you might get  a message saying "conditions not met - DTC clear failed) - so in that instance if you have engine running you might want to try again clearing DTC with engine off but keycard still in the dash slot, and even after that you could still get message that conditions are not met so then you might want to tackle it in 'park mode' so keycard in slot and put in 1st gear (if manual car) or D (drive) if automatic car then while pressing absolutely no pedals a long press on the start button on dash until the words "please remove keycard" - but dont remove keycard, then try clearing DTC's again for the stubbon one(s) that wont clear - if you have a DTC that says "ACTIVE" or "PRESENT" though after you try to clear DTC then its an issue that you or your mechanic will have to investigate more than likely, if its "MEMORIZED" or "----" DTC fault you should be able to clear it with DDT4ALL in a load of cases

shrlnm, 2023-04-14:
It's no matter how to clear, by ddt4all or by pyren.
The response for clearing command was 54 

> Clearing DTC information
> Sending ELM request :14FFFFFF
> ELM response : 54
> Clear DTC successfully done

it means that clearing procedure done their job. But just after that the ECU made 29 unconfirmed DTCs because it had no ability to check and confirm them.

> Sending ELM request :1902FF
> Bad ELM response : WRONG RESPONSE
> Unknown(5902B9
> A0033810
> A0101410
> A03A1410
> A03B1410
> A03C1410
> A03D1410
> A03E1410
> A03F1410
> A0401410
> A0411410
> A0429610
> A0500110
> A0510410 [...]

The only advantage of the pyren is that, thanks to the CLIP database, it can distinguish real problems from unconfirmed DTC

KarelSvo, 2023-04-14:
> Here are the results :

Scan again with engine ON and vehicle selection X98 - CLIO IV.

Trickykid1978, 2023-04-15:
> ehicles and some ecus slept like the engine.
> Scan again with engine ON and vehicle selection X98 - CLIO IV.

Thanks ...what i didn't understand is why vehicle selection plays any role when you use the magnifying glass to find ecus ? (i had the impression that when you use the magnifying glass, the drop down menu of vehicle selection is not taken into consideration).

KarelSvo, 2023-04-15:
If you do not limit the search to your vehicle type, Ecus with (not perfect match) status are also suggested by other vehicle types with similar software versions.
That happened to you with the EPS ECU.

Trickykid1978, 2023-04-15:
> h (not perfect match) status are also suggested by other vehicle types with similar software versions.
> That happened to you with the EPS ECU.

Noted, i will try it as you suggested. From vehicle selection , i was choosing Clio IV-PH2 which is in deed my car ...i will now choose plain CLIO IV to see what it finds

Trickykid1978, 2023-04-15:
> If you do not limit the search to your vehicle type, Ecus with (not perfect match) status are also suggested by other vehicle types with similar software versions. That happened to you with the EPS ECU.
> 
> /

I tried what you suggested but system returned exactly the same ecus as before (4 greens+3 reds = 7 ecus) ... i tried with engine on/off/Clio IV/ Clio IV-PH2 + magnifying glass ....i then scanned with peyren (for the first time) and it found 12 ecus (instead of 4 greens+3 reds = 7 ecus in total from ddt4all) . I compared some of the ecu names and they appeared different to the ones identified by ddt4all ...

AndyIreland, 2023-04-15:
I very rarely put my car into the box - as soon as DDT4ALL starts I click the magnifying glass then select CAN then let it search to 100% , then get green and a few reds. But when I do select my car in the box then press the magnifying search then it goes to 100% and does exactly the same thing and same results , greens and reds - so personally (and  could be wrong) I think it seems to make no difference at all , so I just use the magnifying glass all the time now, only selecting / putting my car into the box when I want to manually change the settings on a particular ECU that (for some reason) the ECU did not show up at all when i searched with the magnifying glass. 

Take this example - when I wanted to enable Android Auto on my Renault Zoe EV I done a search with the magnifying glass in the results box it found radio 2012 ECU but it did not find the MFD ECU (which I needed to enable Android Auto) so I had to add the MFD ECU manually to enable AA  - but I have done a search on other peoples cars with the magnifying glass and it HAS found the MFD ECU automatically. 

Another example: - if I do a search for the TCU on my car it does not find the DCM TCU automatically in DDT4ALL - if I want to change TCU settings I have to manually select the DCM out of the list ..... yet I have done on other people's Renaults and just used the magnifying glass and its found their DCM TCU with no problem and added it to the list. 

I sometimes find that if some people are having trouble finding ECU's is to increase the CAN Timeout on the bottom bar to either 400 or 600 /800ms and experiment with that, then do a search with the magnifying glass. It will take longer to get to 100% but at least it might find more ECU's possibly. 

I do like that with PyREN (apart from you can make a dump of your ECU's) that it seems to find  all the ECU's and puts them in the list plus even seems and feels like there are more settings and i like to use that to clear DTC's for each ECU over DDT4ALL on the whole

Trickykid1978, 2023-04-15:
@AndyIreland 

How many ecus you are missing in ddt4all compared to peyren ?   In most youtube videos online, i see most people don't even bother using the magnifying glass at all ... they just load up their car from the drop down menu and just select the most recent ecu without checking for green and reds etc ...

AndyIreland, 2023-04-15:
> @AndyIreland
>
> How many ecus you are missing in ddt4all compared to peyren ? In most youtube videos online, i see most people don't even bother using the magnifying glass at all ... they just load up [...]

I have just been out to my car now - engine on, 600ms CAN timeout, through the rewired cable to read CAN-HI/rlink radio too. This is what DDT4ALL finds:

[image]

[image]

KarelSvo, 2023-04-15:
> I very rarely put my car into the box - as soon as DDT4ALL starts I click the magnifying glass then select CAN then let it search to 100% , then get green and a few reds. But when I do select my car [...]

The few Renault electric models have very few Ecu siblings in different vehicle models, in contrast to the many vehicles with combustion engines. So the risk of wrong ecus is significantly lower when scanning without vehicle selection.

AndyIreland, 2023-04-15:
always done the same on the wife's Renault Clio Mk4 Petrol as well, always have chose the magnifying glass as a first option and let it find the ECU's i want to change that way

Trickykid1978, 2023-04-16:
Well for some reason, ddt4all finds only 7 ecus when i use the magnifying glass . That's the max it returns back regardless if engine is on or not etc. Apparently my car has more ecus ... do you think is it a matter of database/software used, obd port fault or obd scanner fault ?

KarelSvo, 2023-04-16:
Do you get a connection to the engine ecu if you select EDC17C84 directly?

Trickykid1978, 2023-04-25:
Hello, 

As you can see here https://github.com/cedricp/ddt4all/issues/785#issuecomment-1522170873  , i finally chose the injection ecu which was never found as green or red ecu in ddt4all scans ...i just chose it from the drop down menu of my model ... and it worked fine ... after i disabled start stop , i took it for a ride, start/stopped engine manually several times and at the end, i scanned car for errors with Pyren without any findings ...

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif