cedricp/ddt4all #812: Megane IV Rlink2 downgrade failed 3.3x to 2.2x

Aviv5510, 2023-04-22:
Hello, i tryied to downgrade my Rlink 2 system and failed.
screen went black in the during the downgrade (unofficial file) and stuck like this ever since.
no touchscreen, no radio,  no a\c, absoulutly nothing working.
i have tryied and phyton scripts and still no luck.
any help please ?

AlMeu, 2023-04-22:
Your unit is dead and you nned replace with another.
The problem is all these units installed in cars, are locked to run only in that car.
If you want a brand new unit or old one unlocked, leave an email for contact.

rttr67, 2023-04-22:
I can help  email for contact.

Aviv5510, 2023-04-23:
Avivpele@gmail.com

Thank you 🙏🏼

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif