cedricp/ddt4all #813: Bose Wiring Diagram and ddt4all Activation

createemir, 2023-04-24:
Hello everyone,

I need Bose amplifier wiring diagram for retrofit to R-Link 2 unit. Also, I am looking for amplifier activation on ddt4all. Anybody knows how can I do?

Thanks

HK-Zoe, 2023-05-18:
For what car it is?

createemir, 2023-05-18:
> For what car it is?

Renault Megane IV, phase 1.

HK-Zoe, 2023-05-18:
> > For what car it is?
> 
> Renault Megane IV, phase 1.

Open DDT4ALL and follow these steps:
1 - Choose the car model - XFB MEGANE IV;
2 - ECU name - Navigation - A-IVI v0.9;
3 - Screens - Configuration - System_Information Configuration;
4 - Sound Management - Change Audio_configuration to Premium;
5 - Sound Management - Change Output_information to your Bose amplifier.

[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

createemir, 2023-07-29:
> Open DDT4ALL and follow these steps: 1 - Choose the car model - XFB MEGANE IV; 2 - ECU name - Navigation - A-IVI v0.9; 3 - Screens - Configuration - System_Information Configuration; 4 - Sound [...]

Hi, thanks but this is for Phase 2.

Here are the settings for Phase 1:

1 - Choose the car model - XFB MEGANE IV;
2 - ECU name - Audio- RadNav 3.X v3.3;
3 - Screens - Configuration -ECU configuration - Part I;
4 - ECU - LOUDSPEAKERS - Change Premium Amplifier to Fitted;
5 - ECU - LOUDSPEAKERS - Change Rear tweeters to Not Fitted;
[image]

1 - Choose the car model - XFB MEGANE IV;
2 - ECU name - Audio- RadNav 3.X v3.3;
3 - Screens - Configuration -ECU configuration - Part II;
4 - ADAS - ADAS PRESENCE - Change Sound configuration to AudioBox Ampli;
[image]

AndyIreland, 2023-07-29:
@createemir  - have you got the specially wired cable so that you can read the multimedia through the OBD2 port of the car?

createemir, 2023-07-30:
> @createemir - have you got the specially wired cable so that you can read the multimedia through the OBD2 port of the car?

I have modified my Bluetooth ELM327 V1.5 device (pik18f25k80). Here is the guide that I used:
https://elm327scantool.wordpress.com/2018/11/03/modify-elm327-bluetooth-to-diagnose-renault/

For Megane IV Phase 1, you can reach these ECUs on multimedia CAN line:
-AAU(Audio Amplifier Unit): Configuration and test of Bose amplifier
-Central Panel: Configuration and test of ambient lighting etc.

behtjes, 2024-01-30:
is it possible to set the unite to "none bose" even if you have a bose edition so you get more audio options available (like a equalizer?)

So set audio_configuration to 'classic' and output_configuration to internal amp for example? Or will it get no sound?

createemir, 2024-01-30:
> is it possible to set the unite to "none bose" even if you have a bose edition so you get more audio options available (like a equalizer?)
> 
> So set audio_configuration to 'classic' and output_configuration to internal amp for example? Or doesn't this work?

Could you please give some details about Bose on your car? Do you have subwoofer in spare tire?

To answer your questions, of course it can, but you have to make some wiring. Let me explain how two system work on Phase 1. The options you are talking for Phase 2, which is Easy Link, and it should be containing equalizer settings even with Bose. I will be explaining for R-Link 2 unit.

Summary of classical system:
4 channel R-Link 2 outputs -> Speakers(FR, FL, RR, RL).

Detailed classical system:
R-Link 2 unit have 3 main big connectors on unit. These are 12 pin, 32 pin and 24 pin. Other small connectors for GPS, USB, Radio antenna etc. The 12 pin connector has 10 cables on it. 2 for 12V and GND, and remaining 4 pairs are for 4 channel high level outputs, which means outputs that can be directly connected to speakers. 

Summary of Bose system:
4 channel R-Link 2 outputs -> Bose amplifier inputs — Sound is processed in amplifier -> 5+1 channel outputs -> Speakers(FR, FL, RR, RL, Center, Subwoofer).

Detailed Bose system:
R-Link 2 unit has 4 audio channel outputs on 12 pin connector. On Bose setup, 4 pair of cables are directly connected to Bose amplifier. When Premium Amplifier fitted is selected, these channels delivers differential audio signals to Bose amplifier as input. DSP in the amplifier processes the sound, adds extra 2 channel, that is for center speaker and subwoofer, and sends amplified sound signals to speakers. So, R-Link 2 unit sent audio signals to amplifier, they are processed and sent to speakers. R-Link 2 outputs do not deliver analog signals on external amplifier mode, so output voltage of the R-Link 2 unit does not change with the volume. It is controlled with amplifier. Sound increase and decrease commands are delivered by Multimedia CAN lines to amplifier. 

As a result, if you select Premium Amplifier to Not Fitted, you will have equalizer settings but no sound will be coming from speakers. You have to connect your amplifier outputs with corresponding cables on R-Link 2 outputs, which means bypassing amplifier.

behtjes, 2024-01-31:
I have a subwoofer in spare tire!

Your story makes sense. I don't want to rewire my whole car just to get a equalizer haha. It just sucks that there are no other ways to do this.

One question left: is it possible to adjust the output volume then? The volume on steps are quite high, so on level 5 it's already loud, I'll never get to level 20. So if it's possible to limit the output volume bij 50% or so that would be nice.

Also disable the speed volume (auto adjusting the volume when driving or standing still annoys me)

AndyIreland, 2024-01-31:
> I have a subwoofer in spare tire!
> 
> Your story makes sense. I don't want to rewire my whole car just to get a equalizer haha. It just sucks that there are no other ways to do this.
> 
> One question left: is it possible to adjust the output volume then (in the amplifier?)? The volume steps are quite high, so on level 5 it's already loud, I'll never get to level 20. So if it's possible to limit the output volume bij 50% or so that would be nice.
> 
> Also disable the speed volume (auto adjusting the volume when driving or standing still annoys me)

Have you got a 'LOUDNESS' setting in your menu? - of you knock off that it will quieten it down more at low levels - in the settings as well you should have a Volume speed dependant setting as well to reduce it or disable

behtjes, 2024-01-31:
no there is no loudness or volume speed dependant setting. That's the problem with bose. But maybe it's accessible via DDT4all?

createemir, 2024-02-01:
> I have a subwoofer in spare tire!
>
> Your story makes sense. I don't want to rewire my whole car just to get a equalizer haha. It just sucks that there are no other ways to do this.
>
> One question [...]

I also do not recommend if you are happy with Bose. On video below, volume is nearly 50% and it is not possible to listen at this volume level:
https://www.youtube.com/watch?v=Kroa4J2_3MA

Are you happy with the low frequencies in your car? I made retrofit before, but even bass was +5 in equalizer, there was nearly no bass on low volume levels. I have to turn volume up nearly 40% to hear bass.

Unfortunately there is no option to auto adjust of volume with speed, as I know. Also, there's no loudness setting in R-Link 2. I also couldn't see any option on ddt4all for auto volume adjust.
[image]

behtjes, 2024-02-01:
Thanks for the info! Then i'll live with it. It has good sound but it would be nice if there where some options to tweak.

createemir, 2024-02-01:
> Thanks for the info! Then i'll live with it. It has good sound but it would be nice if there where some options to tweak.

You're welcome. Is it possible to share your acoustic parameters? I tried most of the things but couldn't solve my problems with low frequencies. Only the thing I couldn't try is changing acoustic parameters, because I couldn't find one with fabricationally Bose car.

If you wish, you can find your parameters from this part:
[image]

AndyIreland, 2024-02-01:
> I also do not recommend if you are happy with Bose. On video below, volume is nearly 50% and it is not possible to listen at this volume level: https://www.youtube.com/watch?v=Kroa4J2_3MA
>
> Are you [...]

well if you wanted to see if you could change loudness and speed volume settings you would do it from the Navigation in the list of ECU's anyway , not the premium audio ecu ?

behtjes, 2024-02-01:
> You're welcome. Is it possible to share your acoustic parameters? I tried most of the things but couldn't solve my problems with low frequencies. Only the thing I couldn't try is changing acoustic [...]

I dont have a module for ddt4all. I was just looking for the possibilities. If there were some, i would buy a module.

createemir, 2024-02-02:
> well if you wanted to see if you could change loudness and speed volume settings you would do it from the Navigation in the list of ECU's anyway , not the premium audio ecu ?

Unfortunately, if you have Bose equipped, it is not possible to activate settings such as Loudness, 3 band equalizer etc. on neither ECU list nor R-Link 2. 

For Bose cars, R-Link 2 only provides audio data to Bose amplifier, and rest is job of amplifier. You are only able to adjust Bass and Treble.

For non-Bose cars, R-Link 2 runs with Arkamys based sound processing, which extra options for sound.

bistras2010, 2026-03-28:
> [@createemir](https://github.com/createemir) Hi. Were you able to find the wiring diagram for the Bose amplifier or pinout connectors ?

[BOSE Renault AMP pinout.pdf](https://github.com/user-attachments/files/26325763/BOSE.Renault.AMP.pinout.pdf)

bistras2010, 2026-03-28:
> Please make a (video) tutorial for me (and others) :-)

What kind of totorial you need? How to connect BOSE amplifier to your RLink 2 ? or how to configurate with DDt4all?

cesyouelnino, 2026-09-16:
Hi I have Laguna 3 with system Bose I installed rlink1 but I donte have sound