cedricp/ddt4all #814: Zoe Q210 - Not the basic "Battery charging impossible" - how to fix it?

HK-Zoe, 2023-04-25:
Hello all! I own Zoe Q210 but I have problem with the car. I took the car to service to have the sway bar bushings replaced (everything else was OK). I don't know what the technicians are doing but...

All this is after the replacing the sway bar bushings. First I got message "STOP Electric Failure DANGER" but after lock the car it disappeared. Less than 5 minutes after I lock the car, I get message "Battery charging impossible" and I can't start the car. After I clear EVC with CanZE the Zoe works perfectly fine until I lock the car again. 5 minutes and then I get the same message "Battery charging impossible". 

The errors are:
DF058 (DTC 048373) - TRACTION BATTERY RELAY CIRCUIT
DF018 (DTC 1525F3) - CONSISTENT MULTIPLE SIGNALS FOR CC/SL

[Screenshot of the errors.](https://i.ibb.co/hddwCct/errors.webp)

Does anyone have idea what could be the problem and is there a way to fix it temporary with ddt4all?

I tried with fully charged 12V battery and a brand new one - the issue is still here.

[Video of the issue.](https://www.facebook.com/plugins/video.php?height=314&href=https%3A%2F%2Fwww.facebook.com%2Fhristiyan.d.kolev%2Fvideos%2F175844751954387%2F%3Fidorvanity%3D695124514257075&show_text=false&width=560&t=0)

P.S. I have an appointment for diagnostics at the official dealer on 04.05 (this was the soonest available). But until then I want to try anything else.

AndyIreland, 2023-04-25:
CC is Cruise control - SL is speed limiter. 

Do you think you are comfortable enough with  locking out the zoe to see if this resets any faults? - it might or might not work.

So with drivers door open wait for car to sleep (display and lights go off) then undo the 12v negative battery terminal under the bonnet . 

then remove the cover under the drivers mat covering the lockout (RHD Zoe) and unlock the lockout (you dont have to take it straight out), it will disconnect the 380v DC system . 

so with the 12v system and the 380v system totally disconnected this could be enough to drain the power out of the ECU's - try and leave them disconnected for an hour or 2 or overnight. 

when ready re-connect the orange 380v dc lockout and then the negative terminal onto the 12v battery in that order . 

You more than likely will have to put a code into your Zoe Rlink radionav to get it working again . You can find the number written at the back of your owners book that come with the car or install the my renault phone app and get the radio code from there. 

its worth a try , it does not cost you nothing . Dont do it if you are not comfortable - renault main dealers normally have 500volt DC protective gloves on and PPC gear and safety googles on to perform a lock out of the zoe.  - I have done it loads of times with my Zoe ... and I am still here and have not electrocuted myself yet ,  haha

AndyIreland, 2023-04-25:
ah right at least thats one other thing you have tried. 

Maybe have a go at PyREN and see what DTC errors come up on that as 'memorized' and 'PRESENT'  ?

shrlnm, 2023-04-25:
@HK-Zoe It's better to use qpyren3 and look inside these DTCs
Also look in documentation for description of these DTCs, especially about the first one DTC 048373

AndyIreland, 2023-04-25:
I might sound like I am grabbing at straws sorry, but I think I have heard of a few instances where the relays have 'fused' together on the (could be PEB board or EVC board) before when they have been charged on high speed chargers - it could be that , ... but its just a guess I am not a mechanic or anything .

HK-Zoe, 2023-04-29:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Furtif, 2023-06-02:
i think question is in https://github.com/cedricp/ddt4all/discussions/835 have fun

closed 2023-06-02 by Furtif

KarelSvo, 2023-06-04:
> The errors are: DF058 (DTC 048373) - TRACTION BATTERY RELAY CIRCUIT DF018 (DTC 1525F3) - CONSISTENT MULTIPLE SIGNALS FOR CC/SL

> P.S. I have an appointment for diagnostics at the official dealer on 04.05 (this was the soonest available). But until then I want to try anything else.

How was your problem solved?

RPAuto, 2024-04-18:
i have same problem
RACTION BATTERY RELAY CIRCUIT DF018 (DTC 1525F3)
if i clear error car run when i stop for 5m error can back again.
can someone help me?

HK-Zoe, 2024-04-19:
> i have same problem RACTION BATTERY RELAY CIRCUIT DF018 (DTC 1525F3) if i clear error car run when i stop for 5m error can back again. can someone help me?

Im driving the Zoe with this problem more than a year. If you find any solution, please share it with us.
How did you get the error for first time?

matzel82, 2024-04-19:
> Im driving the Zoe with this problem more than a year. If you find any solution, please share it with us. How did you get the error for first time?

Do want give me your VIN ? I want take a look 👀

HK-Zoe, 2024-04-19:
> Do want give me your VIN ? I want take a look 👀

Send me a email (my email address is in my bio) and I will give you my VIN.

KarelSvo, 2024-04-20:
> i have same problem RACTION BATTERY RELAY CIRCUIT 

[Zoe_PH1.htm.zip](https://github.com/cedricp/ddt4all/files/15047481/Zoe_PH1.htm.zip)

matzel82, 2024-04-20:
That’s not the correct code. It’s from the speedometer….

matzel82, 2024-04-20:
I have found it…i send you a translation per email

KarelSvo, 2024-04-20:
> That’s not the correct code. It’s from the speedometer….

Then give the correct code from @RPAuto,  whose answer I quoted.

HK-Zoe, 2024-04-20:
> I have found it…i send you a translation per email

Okay, i will wait for the email. Thanks!

matzel82, 2024-04-20:
The resistance of the exciter rotor and the brushes in the motor must be measured. This work should be carried out exclusively by a specialised workshop. This is the indication of the error.
[image]

matzel82, 2024-04-20:
ATTENTION: ⚠️ This work is dangerous and may only be carried out by specialist personnel, an activation of the high-voltage circuit is mandatory. There is a danger to life in case of non-compliance ⚠️