cedricp/ddt4all #815: Renault Megane 3 BOSE - UNCONFIGURED RADIO

LemurMort, 2023-04-25:
Hello all!
I experienced a rather unpleasant problem regarding my radio. 
It all started with modifying an OBD cable so I could enable AA. I enabled it succesfully but the curiosity got the best of me.
Using DDT4ALL autoscan found these ECUs:
[image]

So after enabling the AA I went to radio and saw there's a loudspeaker test. I started it, it worked but could not stop it no matter what I did.
So I thought maybe Configuration reset will stop it since it was a bit late and the noise was getting irritant:
[image]

Well, it didn't.
I started to panic a bit and the only way to stop the noise was to disconnect the battery.
After reconnecting it, the radio doesn't work. After turning the ignition ON a yellow triangle appears and a message Unconfigured radio.

I tried to reset the Rlink by pressing home button 5 times - nothing. I transferred the last Rlink update from USB to a new SD card and tried to reset it again - nothing.

I inserted VIN and CRC number into radio ECU and still nothing.

So, the radio doesn't work, parking sensors don't work (visual and auditive aid), rear camera doesn't work.

Should I do a reset on MFD and which kind of reset? 

Can anyone help? If you need any screenshots, please let me know.
Thanks in advance.

BTW I think this is my last time I'm programming something :D

AndyIreland, 2023-04-25:
haha - dont worry dont let it put you off we have all been there well lots (including me anyway and a lot of others) with the dreaded EOL and triangle - I fixed mine at the time at home with this  file on a formatted SD card (not map card)  and fixed it that way ..... but I have a Renault Zoe EV without the BIC module and i think when the reset config is done on a car with a BIC I think it does something to mess up the settings with that so I dont know if it will work on yours . but it wont hurt trying if you like: 
https://we.tl/t-fMfgYXpaE6

did you by any chance use PyREN and back up a 'dump' before making any changes by the way? - maybe you didnt but if you did if you restore the dump file that might get it working too.

LemurMort, 2023-04-25:
Andy, thank you for the file. I'm downloading it as I type this.
I'll try to repeat the process with that file (as I mentioned before, I tried it with my own update (around 500MB) and even with some similar file found on a russian forum (around 2GB).

Unfortunately, I don't have PyREN nor CLIP. only DDT4ALL. Not having a backup wasn't a smart thing to do at all.

We'll see if this will work.

AndyIreland, 2023-04-25:
hope it does

LemurMort, 2023-04-25:
Sorry, it didn't work.
Upon reset the TomTom logo appears and then the dreaded triangle and a message.
[image]

The buttons on radio are illuminated, however. I don't rememember if that was the case before all this hell took place.
[image]

AndyIreland, 2023-04-25:
how much have you got filled out in the ddt4all mfd settings?

LemurMort, 2023-04-26:
Nothing, I think those setings are not messed up.
In radio ecu I wrote VIN and CRC, nothing else.

AndyIreland, 2023-04-26:
have you checked both radio and MFD settings?

shrlnm, 2023-04-26:
@LemurMort  have you seen my message to you in https://github.com/cedricp/ddt4all/issues/717 ?

AndyIreland, 2023-04-26:
@LemurMort  i would need to look to look at your settings in person for the radio and the MFD and I could do over the internet if you have a windows laptop hooked up to the internet and a Bluetooth or USB ELM327 dongle and see if there is anything I can see odd

LemurMort, 2023-04-27:
> @LemurMort have you seen my message to you in #717 ?

Sorry, I missed that somehow. Thanks! I found PyREN, installed it and now I'll see if I can connect to the car. 
I'll post screenshots.

LemurMort, 2023-04-27:
I'm a total newb. I got PyREN184, can't get the version 3 from this page to work.
With that one I did connect to the car.
[image]

I made DUMP files, if that can help (two by each, I don't knowif there's any difference bethween each):
[1682622012_Radio_2012_NoDAB_v2.0_20151012T115515.txt](https://github.com/cedricp/ddt4all/files/11347250/1682622012_Radio_2012_NoDAB_v2.0_20151012T115515.txt)
[1682621931_Radio_2012_NoDAB_v2.0_20151012T115515.txt](https://github.com/cedricp/ddt4all/files/11347251/1682621931_Radio_2012_NoDAB_v2.0_20151012T115515.txt)

[1682622139_BIC_A2_A3_v24_20180620T133153.txt](https://github.com/cedricp/ddt4all/files/11347253/1682622139_BIC_A2_A3_v24_20180620T133153.txt)
[1682622118_BIC_A2_A3_v24_20180620T133153.txt](https://github.com/cedricp/ddt4all/files/11347254/1682622118_BIC_A2_A3_v24_20180620T133153.txt)

Also, here are screenshots I made with DDT4ALL:
RADIO 12 - ECU CONFIGURATION
[image]
[image]
[image]

MFD 12 - ECU CONFIGURATION
[image]
[image]

MFD 13 - ECU CONFIGURATION
[image]
[image]

MFD 14 - ECU CONFIGURATION
[image]
[image]

@shrlnm I didn't find the option ACI in radio block. Is this because of the PyREN version or? Unfortunately I couldn't follow those instructions from drive.ru.

AndyIreland, 2023-04-27:
I dont know if any bearing this has @LemurMort  on anything but you have a CRC of 1BF4 (which is correct) on your MFD ecu settings and a CRC of 3851 in your Radio ECU settings????

LemurMort, 2023-04-28:
I did notice that yesterday but the crc I put into radio is from app My Renault. 
1BF4 is hex, 3851 is decimal; the two have nothing in common 🤷

shrlnm, 2023-04-28:
@LemurMort the right CRC for your VIN is 1BF4 but not sure that it does matter.
The ACI option is absent in menu because of version. It is only in pyren3.
Please send me a dump from the COMMUNICATIONS UNIT  too (MFD).

LemurMort, 2023-04-28:
@shrlnm I made DUMP files by getting into those blocks then into DDT screens, then some .xml file would be listed and then by pressing 'enter' I'd get into it and export DUMP files. However, on this third ECU no .xml files were listed and by pressing 'enter' I'd roll a step back in menu 😞 
I'll make screenshots this afternoon.

AndyIreland, 2023-04-28:
> @shrlnm I made DUMP files by getting into those blocks then into DDT screens, then some .xml file would be listed and then by pressing 'enter' I'd get into it and export DUMP files. However, on this [...]

here is the mfd xml for 5.4 
https://we.tl/t-HbneNBqAZs

- put it in your ecus folder of pyren and when it says what ecu dont choose blank, write the exact name of the file there and then press enter

LemurMort, 2023-04-28:
Andy, may I ask how did you get that file?
Is it a generic file or is it based on my car's VIN?

AndyIreland, 2023-04-28:
out of my ecus file in my program folder. 
i think i might have originally downloaded the larger ecu files from (possibly up on here) called 'larger ECU Zip

shrlnm, 2023-04-28:
> whats the ACI for
ACI is a option in ECU menu of pyren3

ECU : 11339  FG7610684.XML

```
1  - DE : Device errors
2  - ID : Identifications
3  - SY : System state
4  - SP : System parameters
5  - LC : System configuration
6  - AC : Executing tests
7  - CF : Changing configuration
8  - VP : VIN programming
9  - RZ : Resets
10 - FAV : Favourite Parameters
11 - PRA : Parameters list
12 - ETA : States list
13 - IDA : Identifications list
14 - DDT : DDT screens
**15 - ACI : Auto Config Info**
Q  - <Up>

```

It's for getting references for generation proper autoconfiguration macro

LemurMort, 2023-04-28:
Guys, do you think I could do some damage if I load .xml from @AndyIreland ?

Reading explanation from @shrlnm I'm leaning more to direction of ACI. If I get version 3 to work...

AndyIreland, 2023-04-28:
@LemurMort - the xml is not a dump .txt file or a macro file - it wont cure your problem its just the xml file in the database of the ECU's folder for the DDT screens in PyREN

AndyIreland, 2023-04-28:
> ECU : 11339 FG7610684.XML
>
> ```
> 1  - DE : Device errors
> 2  - ID : Identifications
> 3  - SY : System state
> 4  - SP : System parameters
> 5  - LC : System configuration
> 6  - AC : Executing tests
> 7  - CF : [...]

so is the ACI in the actual xml file then or the pyren3 program?  @shrlnm

LemurMort, 2023-04-28:
OK, I loaded Andy's .xml and here are DUMP files from MFD and also again Radio and BIC (so they're all together in one post). 
[1682711838_MFD_v5.4_20190228T163812.txt](https://github.com/cedricp/ddt4all/files/11356931/1682711838_MFD_v5.4_20190228T163812.txt)
[1682711793_MFD_v5.4_20190228T163812.txt](https://github.com/cedricp/ddt4all/files/11356932/1682711793_MFD_v5.4_20190228T163812.txt)
[1682711780_11217.txt](https://github.com/cedricp/ddt4all/files/11356933/1682711780_11217.txt)

[1682622012_Radio_2012_NoDAB_v2.0_20151012T115515.txt](https://github.com/cedricp/ddt4all/files/11356938/1682622012_Radio_2012_NoDAB_v2.0_20151012T115515.txt)
[1682621931_Radio_2012_NoDAB_v2.0_20151012T115515.txt](https://github.com/cedricp/ddt4all/files/11356939/1682621931_Radio_2012_NoDAB_v2.0_20151012T115515.txt)
[1682621926_11193.txt](https://github.com/cedricp/ddt4all/files/11356940/1682621926_11193.txt)

[1682622139_BIC_A2_A3_v24_20180620T133153.txt](https://github.com/cedricp/ddt4all/files/11356942/1682622139_BIC_A2_A3_v24_20180620T133153.txt)
[1682622118_BIC_A2_A3_v24_20180620T133153.txt](https://github.com/cedricp/ddt4all/files/11356943/1682622118_BIC_A2_A3_v24_20180620T133153.txt)
[1682622114_10726.txt](https://github.com/cedricp/ddt4all/files/11356944/1682622114_10726.txt)

shrlnm, 2023-04-28:
> so is the ACI in the actual xml file then or the pyren3 program?

@AndyIreland  ACI function read references from the ECU. Then you may use these data for generating proper autoconfiguration macros

AndyIreland, 2023-04-28:
> > so is the ACI in the actual xml file then or the pyren3 program?
> 
> @AndyIreland ACI function read references from the ECU. Then you may use these data for generating proper autoconfiguration macros

oh right thanks.

You know that .cmd  Macro script - does that work (or is there one to work with ) the rlink1 radio/nav with the EOL triangle 'not configured' problem or does it only work on rlink 2 radio/MFD ?

AndyIreland, 2023-04-28:
> OK, I loaded Andy's .xml and here are DUMP files from MFD and also again Radio and BIC (so they're all together in one post). [...]

errrr ... but they are all dumps / rollback files off your own car where the er radio is all unconfigured with a triangle on your screen!

see what happens is that you do it this way **before** you make any changes on your car's ECU's 
Then you make a change or changes then find you have corrupted your ECU settings and think "not to worry PyREN has made a .txt backup/dump" so then you go into that ECU with Pyren then go into the 'Rollback' setting and select the last .txt dump file that pyren automatically backed up for you (well thats the theory on how it works anyway) 

if your ECU is already corrupt or wrongly configured then let Pyren do a dump .txt file all you are doing is backing up unconfigured/incorrectly configured/corrup ECU settings - so if you rollback you are back to square 1 

you could try when you are in PyREN selecting "prefer to use from xml file" instead of 'Prefer ECU settings' possibly and see if that works. i havent tried it though.

LemurMort, 2023-04-29:
Thanks for the explanation, @AndyIreland . I kind of knew that but as @shrlnm requested earlier I uploaded it. 

There's a great video on yt by user Andrey Amokk about PyREN, macros and ACI. I believe I could solve my case but I can't get bloody PyREN3 to work 😕

shrlnm, 2023-04-29:
Hello @LemurMort 
I made macro for you
https://cloud.mail.ru/public/35nz/wCH7aCuSU
https://cloud.mail.ru/public/eAaY/5Hm7tVbWS
You may implement them by old pyren.  Do not forget to enable logging before implementation and check that no errors will be in the output.

shrlnm, 2023-04-29:
> You know that .cmd Macro script - does that work (or is there one to work with ) the rlink1 radio/nav with the EOL triangle 'not configured' problem or does it only work on rlink 2 radio/MFD ?

Hello @AndyIreland 
With macro scripts you may repair RLink1//MFD as well as Rlink2 (it has no MFD). 
Actually you may repair almost any ECU by this method.
Have you read my articles?
https://www.drive2.ru/b/643019689764615344/
https://www.drive2.ru/b/643355315688994881/

LemurMort, 2023-04-29:
@shrlnm thank you! I hope I'll try it today.

I'm not profficient in these things and this in combination with lack of time makes me rather not to make any hastly moves again (like resetting the radio configuration 😃).

It takes me some time to explore how to do something. As you wrote to ONLY install Python 3.xx (seems like a simple step to do), I'd be probably googling and whatnot sporadically for days to find that out.

Thanks once more! 🙂

LemurMort, 2023-04-30:
Ahhhh, look what time it is (02:09). Where do I start...

So, I removed previous version of PyREN and I followed the yt video, installed python 2.7.11, followed all the steps.
I even made my car's documentation. It's in russian. Any chance to get it in english?
Some problem with ELM but I finally connected to the CAN2 bus.

**Problem no.1**
I used this DDT db:
[image]

Again, both Radio and BIC have their .xml files, MFD doesn't:
[image]

I go to folder 'ecus' in DDT db archive so I could paste .xml from @AndyIreland but the file is already there. 
I put Andy's file and it's identical:
[image]

Still, no MFD ecu is offered in DDT screen.

**Problem no.2**
I copied macros from @shrlnm into corresponding folder:
[image]

I click on macro button on start screen...
[image]

... And this appears:
[image]

Do you know what's happening?

shrlnm, 2023-04-30:
Hello @LemurMort 
Let's begin from the end.  
Our main goal is to run macros I made for you. You did everything almost right but ticked "Data Logging/csv" option. It don't work in macro mode. Just disable it and repeat
[image]
This option is only for pyren mode.

All other problems are related to outdated pyren. Recently I fixed the problem with MFD modules. I don't know why but MFD modules in CLIP database have an address '57' but in DDT '58'.

The pyren3 works only with python 3.xx 
https://www.python.org/ftp/python/3.10.11/python-3.10.11-amd64.exe
during installation enable 'Tcl/Tk support' and allow the installer to edit paths 
the pyren3 is there https://gitlab.com/py_ren/pyren/-/archive/pyren3/pyren-pyren3.zip

For generation documentation in english use the pyren_bot with command **/doc <your vin> GB**

LemurMort, 2023-04-30:
Good morning @shrlnm .
Thank you for your instructions.

I installed python 3.10.11.
Do I have to be connected to ELM to run macros?

As for pyren3, i extracted the zip here:
[image]

As for documentation in pyren_bot i got message: 
[image]

shrlnm, 2023-04-30:
@LemurMort You may try a macro in demo mode. In launcher you may enter "--demo" in bottom "Other options" field and press button "Macro"

move files from inside pyren-pyren3 one folder up. main.py and folder pyren3 should be placed next to MTCSAVE.

Also you may try to use pyren_bot once again now.

LemurMort, 2023-05-01:
Good morning @shrlnm and thank you!

I received the files from pyren_bot yesterday. Interesting stuff.
I didn't know there are dump files. 
If I understand correctly; I can now restore my system with these dump files OR with macros?
In dump files I received I noticed this difference in MFD versions:
[image]
[image]
If I were to restore via dump, would this be a problem?
- - - - - - - - -
Starting macros in demo mode works,
I notice a difference in filenames between macros you posted earlier and the ones I got from bot.
[image]
can2_10_... was identical, but can2_76_... is different. I don't think this matters.

Inside is all identical except at the end:
[image]
Should I use the ones **with #** or **without #**?

 - - - - - - - - - - - - - - - - - - - - - - - - 

I'm sorry if I misunderstood you. 
I copied contents of folder **C:\PyREN\pyren-pyren3** to **C:\PyREN\pyren** but I still don't have the option to select the version pyren3 in launcher after I pressed 'UPDATE' button:
[image]

I downloaded the last version of pyren3 with the mentioned address fixed, I update pyren but still cannot load MFD .xml :/
 - - - - - - - - - - - - 
And regarding the stuff that scares me:
[image]

What are the correct steps I should make?
Can I run any macro first or is there a particular order?
After running the macro will the instructions be written what to do next?

LemurMort, 2023-05-01:
Here's the end result:
[image]

I got Pyren3 to work, ran macros 76: then 10: and reset after every macro and nothing.
So I ran API for MFD and radio, got another set of macros from bot, uncommented last lines, then ran 10: and then 76: and then I heard a beautiful sound from each speaker for maybe 2min. Then I inserted pin (from Renault app, that's not the crc as I thought before) and it works now as it did before.
Made dump files, then activated AA only. I have it in the menu but can't connect phone to it. I'll try with a newer smartphone. 

Thank you all for your help, good will and patience 🙂
@shrlnm your guide now makes a lot of sense to me. At the beggining I didn't understand anything 😃

AndyIreland, 2023-05-01:
glad you got it sorted @LemurMort  and well done for all the help @shrlnm

LemurMort, 2023-05-01:
Thanks @AndyIreland, certainly me too.
As for AA, it was the cable. 
Now that works too.
Hey google...
[image]

AndyIreland, 2023-05-01:
> Thanks @AndyIreland, certainly me too. As for AA, it was the cable. Now that works too. Hey google... [image]

great, I was about to suggest a different cable. Same happened to me as well on my car. once i bought a decent one it worked every time

closed 2023-05-02 by LemurMort

runsren, 2023-10-01:
This thread got my Smart Media System working, thanks.
Special thanks for the Macro files.

stumbaumr, 2024-11-02:
Dear @shrlnm , do you think that this might also work with [my Smart EQ from 04/2019](https://www-smart--emotion-de.translate.goog/forum/thread/3533-mit-ddt4all-rumgespielt-loudspeaker-test-angemacht-und-in-den-panikmodus-verfall/?_x_tr_sl=de&_x_tr_tl=ru&_x_tr_hl=de&_x_tr_pto=wapp)?

shrlnm, 2024-11-03:
Hello @stumbaumr. I not sure that your VIN is in configuration DB. You may try to implement configuration from some other ranault car.