cedricp/ddt4all #821: Talisman 2017 replace R-Link2 screen with used from Koleos. Now ambient light dashboard not working

rick385, 2023-05-04:
The garage (not Renault) replaced my original R-Link2 screen with a used screen because the previous owner had made a lot of scratches on the creen and bezzels. The replacement screen is in near new condition and came from a Koleos.

Everything works fine. 
Except the ambient lighting below the dashboard does not work anymore.

What I now understand is:
- the replacement screen still has the VIN from the Koleos
- my R-Link2 head unit is of course Talisman
- these are in conflict so the Koleos VIN in the new screen has to be updated to my own Talisman VIN
- when the garage read the fault codes they indeed detected permanent errors, also in the doors. The Koleos has ambient light in the doors. The Talisman has only ambient light in the dash. So this is in conflict.

Am I correct in my analysis of the problem? 

I have an appointment with workshop chief of the official Renault dealer. He is experienced with the CLIP.
Can he just update the Koleos VIN to the Talisman VIN with the CLIP and then reconfigure it to Talisman?
Or does he need to "virginize" the screen before this? And then reinstall the R-Link software? I am now running V9.

I am not aiming at fixing this myself, just want some background info so that when I bring my Talisman to the dealer I know what are the options. Sometimes the dealer just says "we cannot fix this, you need a brand new screen"

Lauddt94, 2023-05-04:
@rick385 recently I had the same issue with my Megane, I updated the screen en the ambient light was not working in every section. My solution was a modified ELM adapter which gives access to the Media CAN, there is a menu in DDT4ALL where you can adjust the ambient lighting state where all possible different zones are selectable.

rick385, 2023-05-04:
@Lauddt94 : Thanks for the reply. I read another owner who reported this on drive2.ru..

So you fixed it yourself?
Can you share a screenshot of ddt4all where you corrected this? Did you replace the screen with a screen from the same model Renault?

Lauddt94, 2023-05-04:
I replaced it with a 8,7'' screen, I don't know from what type car it came, probably another type. 

https://megane.com.pl/topic/79434-multisense/page/6/

On this page you see three screenshots off the pages I mentioned.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif