cedricp/ddt4all #822: ddt4all crashes when opens an ECU virgin

AlMeu, 2023-05-09:
I try run ddt4all v1 detects the ECU with VIN empty and when try open it is crashes!!!

[image]

KarelSvo, 2023-05-09:
[ULC_gen4.1_v0.5_20210325T103439.xml.zip](https://github.com/cedricp/ddt4all/files/11434559/ULC_gen4.1_v0.5_20210325T103439.xml.zip)

I don't know where the problem is in ddt4all. I have removed the ReadVIN from Screen 12 in the ECU file.
You can load the file via File/Open XML.

AlMeu, 2023-05-09:
I open your XML and doesn't crash when I click on that tab.
There it is ddt4all the problem. I have even the old version with python 2.7 and in that doesn't crash but don't open that tab when the VIN is empty.

KarelSvo, 2023-05-09:
Perhaps there is still an ecu.zip in the program folder without this file?

AlMeu, 2023-05-09:
I have an ecu.zip and I renamed as ecu______.zip and it reads the same.
As I know also in that ecu.zip contains these.
I will remove ecu.zip and retry once again.

AlMeu, 2023-05-09:
@KarelSvo 
I have tested also without ecu.zip and it is showing same list, even if ecu folder have more versions.

AlMeu, 2023-05-09:
I ddt4all with python 2.7 same errors, but not crashes, don't open that page but keep me the window opened.
[image]

KarelSvo, 2023-05-09:
Does the ULC file appear if you don't select a vehicle type?
Tried ULC VIN in Pyrene mod_ddt?

AlMeu, 2023-05-09:
If I have ecu folder without selecting vehicle the ULC ECU still doesn't appears all.
Then I tried and removed ecu folder and left only ecu.zip and in this way it shows all versions of ECU ULC.
I am using Pyren too, but seams also this gives the same error to this ECU with empty VIN and here even block the scanning!!!!
[image]

KarelSvo, 2023-05-09:
Try this change in ddt4all file ecu.py
[image]

AlMeu, 2023-05-09:
Yes, in this way, doesn't crash anymore. :+1: 
It opens the pages anyway, and it shows empty VIN fields. I hope to let me write in future the VIN there.

KarelSvo, 2023-05-09:
I think so.
You need to find the file and line from the end of the crash message.

AlMeu, 2023-05-09:
I have done, and it works
[image]
Thanks

closed 2023-05-09 by AlMeu

MiqdadAli2002, 2023-05-10:
[image]
Can you help me with this issue please