cedricp/ddt4all #823: Time for automatic re-lock

LemurMort, 2023-05-09:
Hello all!
Curious Lemur here 😃
I have a question regarding duration of automatic re-lock (if after unlocking the car no door is opened the car locks itself again).
Default value is duration in minutes (0, 1, 2...).

I'm curious if decimal values can be put here.
Ie. for 30s should the value be 0.5 or 0,5 (with **dot** or **comma**)?

Thank you!

Furtif, 2023-06-02:
is hard to reply try Community as https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif

KarelSvo, 2023-06-04:
> I'm curious if decimal values can be put here. Ie. for 30s should the value be 0.5 or 0,5 (with **dot** or **comma**)?
> 

If you don't tell the name of the relevant ECU file, it's difficult to give help.

AndyIreland, 2023-06-04:
something like this ECU  I suppose in BCM/UCH:

[image]

This is for BCM95 SW16 V16_1 --- but you may have a different BCM ECU version in your Megane3

LemurMort, 2023-06-04:
Hey guys. My BCM is BCM95_SW25_V1. 1

AndyIreland, 2023-06-04:
> Hey guys. My BCM is BCM95_SW25_V1. 1

so this then- but i could not tell you if using decimal point will work sorry (as in 1.5 means 1 and a half minutes)

[image]

KarelSvo, 2023-06-04:
> I'm curious if decimal values can be put here.

The value sent uses only 1 byte.
1 byte can only represent [whole decimal values between 0 and 255](https://kb.iu.edu/d/afdl).

[image]

LemurMort, 2023-06-04:
Thanks a lot for your help! 🙂