cedricp/ddt4all #824: Renault Zoe Q210 -

HK-Zoe, 2023-05-10:
Hello,
I own Renault Zoe Q210. Is it possible to activate keyless entry and keyless start with ddt4all?

AndyIreland, 2023-05-10:
it possible - but you would have to have all the aerials inside the car, there are loads of them all over the place

HK-Zoe, 2023-05-10:
What are the all aerials that I have to replace and how to activate it after that with ddt4all?

AndyIreland, 2023-05-10:
the handsfree module aerials are in the front doors and behind the handbrake lever, and behind the dash near the cigarette lighter socket and in the boot area. 
If your car did not come with handsfree entry and keyless then none of the aerials will be there in the car (possibly neither the wiring looms for the aerials as well) - be a lot of work to put them all in I would imagine. 

the DDT4all settings would be in the BCM module 

for keyless:
[image]

for handsfree door entry:
[image]

possibly more settings elsewhere

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/discussions

Furtif, 2023-06-02:
it is not a issue

closed 2023-06-02 by Furtif