cedricp/ddt4all #825: Renault Talisman Front camera issue

MiqdadAli2002, 2023-05-10:
Hello
I have an issue with my Renault Talisman 
The issue is with the front camera 
when I scan the errors I find that
(DTCAD0451 COMPUTER CALIBRATION)

KarelSvo, 2023-05-11:
DTCAD0451 front camera : Diagnosis procedure
Performing calibration of the front camera

Force the vehicle's + after ignition feed.

Wait for 5 seconds.

Calibrate the computer.

Note!!!

The targets (Ms. 1994) should be placed in front of the vehicle.

Ensure that the field of vision of the camera is free.

Verify that the protection of the lens is removed.

RENAULT-CLIP-LAUNCH-ACTUATOR Domain:(FRONT CAMERA) CF001-COMPUTER CALIBRATION

Note!!!

If the calibration is not performed correctly, please check that targets are clear and correctly positioned, as well as the camera.

MiqdadAli2002, 2023-05-11:
Is there another way to solve it because I don’t have laser device or targets and no workshop in my region does it

MiqdadAli2002, 2023-05-14:
@shrlnm can you help me please?

shrlnm, 2023-05-14:
@MiqdadAli2002 unfortunately, I don't know how to help with calibaration

MiqdadAli2002, 2023-05-14:
I need file dump please

shrlnm, 2023-05-14:
@MiqdadAli2002 you may generate major parameters by pyren_bot, but not sure that they contains calibrations.

MiqdadAli2002, 2023-05-14:
actually the bot doesn’t work

MiqdadAli2002, 2023-05-14:
dump for front camera because I have many none

shrlnm, 2023-05-14:
@MiqdadAli2002 bot is working. Do not type brackets <> in commands. Just
/vin VF1RFD005JC25xxxxx

shrlnm, 2023-05-14:
Bingo! )

Furtif, 2023-06-02:
for discutions and not issue please uses https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif