cedricp/ddt4all #828: Disable SOS call

Tony2k, 2023-05-18:
I replaced the original radio of the Dacia Spring with a generic one (non-can) and now the writing "check sos call" appears on the panel.  Why does the call have to go through the radio if it is not equipped with GPS or a telephone module?  Is it possible to disable this function with ddt4all?

AndyIreland, 2023-05-18:
could be if you can unlock the security gateway (if its got one) with a token, and then do a search in DDT4all and if it finds 'Clustre' look on settings and see if you can turn off the SOS message on the dash

Tony2k, 2023-05-27:
Do you mean "cluster"?

KarelSvo, 2023-05-28:
TCU DataWrite.Ecall Activation Status
Reg Ecall activated to Reg Ecall deactivated

Tony2k, 2023-05-29:
> TCU DataWrite.Ecall Activation Status Reg Ecall activated change to Reg Ecall deactivated Cluster DataWrite.TCU_EcallPresent_CF With Emergency_Call change to Without Emergency_Call

Can you explain me please where is DataWrite.Ecall Activation Status?

[image]

AndyIreland, 2023-05-29:
in my Renault Zoe EV there is some ecall settings in the rlink1 medianav MFD  settings:  I couldnt say where it is on your dacia spring . Maybe in the easylink radio or whatever you have? 

[image]

KarelSvo, 2023-05-29:
TCU_AIVC_v2.27/CONFIGURATION/Reg Ecall 1/Ecall Activation Status  Line number 6

Tony2k, 2023-05-29:
How solved the problem working on DTC Activation Status:

[image]

closed 2023-05-29 by Tony2k

rxtarcher, 2024-08-03:
Hello, I got exactly the same message "Check SOS Call".

I am trying to go through ddt4all in order to deactivate "Aux-In Info/Tel-Mute-Output" and "Virgin Configuration", but it doesn't work. I press the "Einstein" button in order to activate expert mode and change the value from "Activated" to "Deactivated", I press "Write activation status of DTC" and then "Key On/Off Reset" but nothing happens. If I refresh or close and open ddt4all both options are still "Activated". 
Also, in the "Read" right column, I can't see if an option is "Activated" or "Deactivated", they are all write "NO DATA".

Do you know what do I have to do in order to be able to deactivate both options?

Thank you.