cedricp/ddt4all #829: DTC 926755 no way to erase it

JABICHOCU, 2023-05-26:
As many of you already know, I´ve been trying to inhibit stop-start on my 2013 captur for a while...until lately I managed to successfully do it on my EMM unit.But now I get this error on my Pyren diagnosis, which I have no way to erase...does this mean car is not working properly or is it just informing that  S-S system is  inhibited? [
[image]
](url)

KarelSvo, 2023-05-27:
DTC xxxx55 means not configured

JABICHOCU, 2023-05-27:
Yes...that sounds interesting but out of my knowledge. For other cars you already have "ready to go" aftermarked switches to do what you say, but not for this one (as far as I know)

Furtif, 2023-06-02:
you screen i like is a Pyren ... if my memory is good.. but it is not a issue please ask in https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif