cedricp/ddt4all #830: Unkown Protocol, Protocol not supportet

NathanScott89, 2023-05-28:
In the Database from 11.2021 are ECUs which are not supportet.

A_IVI2_MyF1_v1.56
A_IVI2_ABPT2_v1.53
A_IVI2_MyF2_v1.1

The protocol is unknown. If I want to connect to the device, DDT4All shows the message "protocoll not supportet".
Which protocol is it?

These ECUs are the new Android Automotive Navigation Module from the Megane E-Tech and Austral.

NathanScott89, 2023-05-28:
DDT2000 shows that the protocoll ist DoIP.
Support DDT4All DoIP?

AndyIreland, 2023-05-28:
does your car have Security gateway it being a newer renault? 
If so have you unlocked the gateway with a token before using DDT4ALL?

NathanScott89, 2023-05-28:
Yes it has a security gateway. But I only wanted to read an not write.

If I connect on the CANBUS behind the Gateway  the same message came.
But I think CANBUS ist not correct for DoIP. DoIP need 4 PINs.

The Message came also in the Demo/Testmode.
Works the ELM with DoIP?
Support DDT4All DoIP?
Why is in DDT4All the protocol unknown and DDT2000 shows DoIP?

Furtif, 2023-06-02:
odd as question uses https://github.com/cedricp/ddt4all/discussions  maybe as some users reply it

closed 2023-06-02 by Furtif