cedricp/ddt4all #860: [Crash] - connected or edition mode

watchix, 2023-06-28:
Hello,

I'm having a crash after double-clicking on every object in this left bar :
[image]

Logs of the crash :
[image]

Regards,

KarelSvo, 2023-07-01:
There are some DDT2000 files from old ECUs,  that need to be repaired before use. The ddt2000 BuilderX Editor shows you, what to do.
[image]
[image]

Furtif, 2023-07-04:
it seems not to be a ddt4all error but the ecu in question

I will convert to discussion.