cedricp/ddt4all #864: Can't seem to open any ecu and do anything

drder66, 2023-07-02:
I can open ddt4all, navigate to an ECU, but the moment I try to click on anything to open it,  I get 

Using custom DDT database
1366 loaded ECUs in database.
Traceback (most recent call last):
  File "./main.py", line 1039, in changeECU
    self.set_param_file(ecu_file, ecu_addr, ecu_name, isxml)
  File "./main.py", line 1055, in set_param_file
    self.paramview = parameters.paramWidget(self.scrollview, ecu_file, ecu_addr, ecu_name, self.logview,
  File "/home/der/Downloads/SmartCar/ddt4all-2.1.1/parameters.py", line 241, in __init__
    self.initXML()
  File "/home/der/Downloads/SmartCar/ddt4all-2.1.1/parameters.py", line 796, in initXML
    self.ecurequestsparser = ecu.Ecu_file(xdoc)
  File "/home/der/Downloads/SmartCar/ddt4all-2.1.1/ecu.py", line 901, in __init__
    xdom = xml.dom.minidom.parse(data)
  File "/usr/lib/python3.8/xml/dom/minidom.py", line 1959, in parse
    return expatbuilder.parse(file)
  File "/usr/lib/python3.8/xml/dom/expatbuilder.py", line 913, in parse
    result = builder.parseFile(file)
  File "/usr/lib/python3.8/xml/dom/expatbuilder.py", line 204, in parseFile
    buffer = file.read(16*1024)
AttributeError: 'Element' object has no attribute 'read'
./ddt4all.sh: line 16: 2387823 Aborted                 (core dumped) ./venv/bin/python3 ./main.py

drder66, 2023-07-02:
the ECU file seems to be able to load ok
python -c "import sys, xml.dom.minidom as d; d.parse(sys.argv[1])" MFD_v4.6_20170209T154731.xml 
returns no errors.
However the function gets called again:
Trying to load ECU file ecus/MFD_v4.6_20170209T154731.xml
Trying to load ECU file <DOM Element: xml at 0x7ff8733a65e0>
And the second time round crashes as previously detailed.

drder66, 2023-07-07:
Duplicate of #860

drder66, 2023-07-08:
The problem is a piece of convoluted code that ends up trying to re-open the ECU using a dom handle rather than the ECU file name. To fix the problem, at line 897 in ecu.py:

Replace this code
------------------------
            if isfile:
                if not os.path.exists(data):
                    print("Cannot load ECU file", data)
                    return
            xdom = xml.dom.minidom.parse(data)
            self.xmldoc = xdom.documentElement
-------------------------
With this code:
           if isfile:
                if not os.path.exists(data):
                    print("Cannot load ECU file", data)
                    return
                xdom = xml.dom.minidom.parse(data)
                self.xmldoc = xdom.documentElement
            else:
                self.xmldoc = data
--------------------

drder66, 2023-07-08:
At line 897 in ecu.py
Replace this code:
```

            if isfile:
                if not os.path.exists(data):
                    print("Cannot load ECU file", data)
                    return
            xdom = xml.dom.minidom.parse(data)
            self.xmldoc = xdom.documentElement

```
With this code:
```
           if isfile:
                if not os.path.exists(data):
                    print("Cannot load ECU file", data)
                    return
                xdom = xml.dom.minidom.parse(data)
                self.xmldoc = xdom.documentElement
            else:
                self.xmldoc = data

The problem is that data can be either a file name or an XML DOM reference. Feeding the DOM reference to parse crashes it. When this function is called with a reference, there is no need to parse it again.

Furtif, 2023-07-26:
> At line 897 in ecu.py Replace this code:
>
> ```
>
>             if isfile:
>                 if not os.path.exists(data):
>                     print("Cannot load ECU file", data) [...]

if fine you code make a pr and propose changes.

Chumneywarner, 2023-08-17:
Hi,

I am new to this, so please excuse any ignorance.

I have the same issue as drder66.

I am using Windows 11 on a 64 bit machine & have installed DDT4all 23.06.13-python-3.11.4-64bits.
I have tried using the drder66 fix & have replaced the code as advised (I believe I have done this correctly), but I still have the same issue.

If I uninstall DDT4all 23.06.13-python-3.11.4-64bits & reinstall using DDT4all 23.06.13-python-3.8.6-32bits, all seems to work OK, but as I have a x64 machine, obviously I would prefer to use the x64 version of DDT4all.
Any ideas on a fix for this issue please?

Thanks
Chumneywarner

radevon, 2023-09-27:
> At line 897 in ecu.py Replace this code:
>
> ```
>
>             if isfile:
>                 if not os.path.exists(data):
>                     print("Cannot load ECU file", data) [...]

I tried to use it, and it is fine worked. (windows 7, x64) Thanks!

Chumneywarner, 2023-09-29:
Thanks for your help.I'll try this as soon as I have time.Cheers

Furtif, 2023-11-12:
I think your problem is very strange... it's either file or data

Furtif, 2023-11-12:
propose pr if not ok...

closed 2023-11-12 by Furtif