cedricp/ddt4all #868: Connecting OBDII by WiFi or Bluetooth - does not work with macOS

Vape2Die, 2023-07-18:
Hi.
I installed everything well, when launching the program in macOS - the software does not recognize WiFi and Bluetooth OBDII connections.
While the exact same OBDII plugs DO WORK with Windows 10.
There must be something wrong with the code that prevents the connection.
I made sure all the necessary packages were installed according to requirements.txt

- My OBDII Plugs do support ELM327 and have in them the PIC18F25K80 chip.
- Both plugs are working in the Windows 10 program, so there shouldn't be any issue with my hardware

KarelSvo, 2023-07-18:
https://github.com/cedricp/ddt4all/pull/659/commits/1c68148f3e1eb0102878b4fd20acdad54bd1c401

Vape2Die, 2023-07-18:
Thanks for the quick response @KarelSvo.
I have implemented the change in the code, but now encountering the following error:

[image]

Vape2Die, 2023-07-18:
I have added ignore flag
`    def read(self):
        try:
            byte = b""
            if self.portType == 1:
                import socket
                try:
                    byte = self.hdr.recv(1)
                    print(str(byte))
                except socket.timeout:
                    self.tcp_needs_reconnect = True
                except Exception as e:
                    print(e)
            elif self.portType == 2:
                if self.droid.bluetoothReadReady():
                    byte = self.droid.bluetoothRead(1).result
            else:
                if self.hdr.inWaiting():
                    byte = self.hdr.read()
        except:
            print('*' * 40)
            print('*       ' + _('Connection to ELM was lost'))
            self.connectionStatus = False
            self.close()
            return None
        try:
            return byte.decode("utf-8", 'ignore')  # Ignore undecodable characters (Vape2Die)
        except:
            print("Cannot decode bytes " + str(byte))
            return ""`

So the response in terminal does not flag `cannot decode bytes b'\xfc'` anymore.. but still ddt4all loads with 0 ECUs
I can't understand why it happens, any idea @KarelSvo?
[image]

Vape2Die, 2023-07-19:
@KarelSvo You're PRO!!!
thank you so much! works great now :)

closed 2023-07-19 by Vape2Die