cedricp/ddt4all #872: Problème avec ddt4all

just4545, 2023-07-25:
Bonjour

Je me permet de créer ce ticket afin de vous informer que je rencontre un problème ,j'ai installé votre dernière version de ddt4all -> https://github.com/cedricp/ddt4all/releases/tag/v2.1.1 , une fois installé je lui ai importé la base de donnée jusque la tout est OK.

J'utilise et obdII en WIFI ->https://www.amazon.fr/dp/B01NAQ2878?psc=1&ref=ppx_yo2ov_dt_b_product_details, la connexion avec ddt4all se fait correctement. Si j'utilise ddt4all c'est seulement pour activer l'Android Auto sur un R-link.

Parcontre quand je fais des modifs et que je les applique rien ne se passe sa reste bloquer sur le screen d'en dessous : 

[image]

et ça fait rien après, pourriez m'aider s'il vous plait 

Merci d'avance

AndyIreland, 2023-07-26:
try pressing the 'home' button to the right of the touchscreen for the rlink   5 times and see if it reboots the rlink and then see if AA has installed

Furtif, 2023-07-26:
Pour activer AA je crois que aller dans micom suffit si c'est le même principe que media nav.
Par autre coté je crois que les fonctions multimédia sont accessibles via CAN2 et un OBD2 basic n'a pas accès a ce coté la mis a part en croisé, voir https://github.com/cedricp/ddt4all/wiki#how-to-access-multimedia-can-bus

just4545, 2023-07-26:
> Pour activer AA je crois que aller dans micom suffit si c'est le même principe que media nav. Par autre coté je crois que les fonctions multimédia sont accessibles via CAN2 et un OBD2 basic n'a pas [...]

just4545, 2023-07-26:
Bonjour @Furtif 

Merci d'avoir pris le temps de me répondre , pourrais-tu me dire c'est quoi micom ? car j'ai suivi cette video : https://www.youtube.com/watch?v=ZXObC7j4WtM&pp=ygUeYWN0aXZhdGUgYW5kcm9pZCBhdXRvIGluIHJsaW5r et il utilise bien le obd2 que j'ai acheté mais je comprends pas pourtant le obd2 se connecte bien a mon pc etc.. mais il arrive pas a écrire ce que je lui demande.

Furtif, 2023-07-27:
Avais vous cliquer sur l’icône d’Albert pour passer en mode expert ? 

Pour micom voir https://m.youtube.com/watch?v=5pElphWMa9o

https://www.motorcarsoft.com/viewtopic.php?t=17928

just4545, 2023-07-27:
Bonjour @Furtif 

Oui j'ai bien cliqué sur le bouton d'Albert mais sur le coté droit il y a marqué NO DATA sur tous et quand je fais read all sa ne fait rien

J'ai changé de prise j'utilise cette prise maintenant : https://diagauto-store.fr/products/valise-de-diagnostic-multimarque-ds-150e-en-francais-bt-usb?gclid=EAIaIQobChMIlfLZ252vgAMVSEBBAh1pKwnsEAQYAiABEgKA7vD_BwE   sa devrait quand meme fonctionner avec ddt4all ?

Furtif, 2023-07-27:
> Bonjour @Furtif
>
> Oui j'ai bien cliqué sur le bouton d'Albert mais sur le coté droit il y a marqué NO DATA sur tous et quand je fais read all sa ne fait rien
>
> J'ai changé de prise j'utilise cette [...]

je ne sais pas... mais c'est pas dit normalement si a tous a NO_DATA c'est car vous n'avais pas accès au coté CAN2 comme dit ici https://github.com/cedricp/ddt4all/wiki#how-to-access-multimedia-can-bus pour y avoir acces a ce coté vous faut une sonde Derelek ou Bosh car on 2 modes http://www.derelek.com/nos-produits/ (normalement USB-DIAG 3) ou croiser les pins comme dit dans le wiki du github voir le site russe que explique comment faire.

Votre DDT4All est bon faut just que vous trouvais une solution pour rentrer dans coté multimedia CAN2

Furtif, 2023-07-27:
[image]

comme vous pouvais le voir il y a 2 modes (en gros il croise le pin) mode Diag (multimédia CAN2) e mode véhicule.

just4545, 2023-07-27:
Et vous pensez que avec celui-là va fonctionner : https://www.amazon.fr/dp/B07MQ8GHG3?psc=1&ref=ppx_yo2ov_dt_b_product_details

Furtif, 2023-07-27:
> Et vous pensez que avec celui-là va fonctionner : https://www.amazon.fr/dp/B07MQ8GHG3?psc=1&ref=ppx_yo2ov_dt_b_product_details

Honnêtement je peut pas vous le dire car je ne l'ai pas tester. Désolé.

just4545, 2023-07-27:
ça marche, je vais tester et puis je vous dirai
Merci pour votre aide :)

Furtif, 2023-07-27:
de tout façon vous aller avoir accès que coté véhicule même que ce câble marche, comme je vous ai expliquer pour accès CAN2 y a que 2 solutions soit utiliser la sonde derelec destiné d'origine a DDT2000 (a savoir que ddt4all utilise les database de ddt2000 et est basé sur ce logiciel) soit utiliser un OBD2 comme lui 

[image]

et lui faire une manipulation pour avoir les 2 modes

[image]

voir https://www-drive2-ru.translate.goog/l/468057702971474426/?_x_tr_sl=auto&_x_tr_tl=fr&_x_tr_hl=fr&_x_tr_pto=wapp (je mise le site dans google translate pour lire un peut mieux).

just4545, 2023-07-27:
oui alors celui que je vous ai montré : https://www.amazon.fr/dp/B07MQ8GHG3/ref=pe_27091421_487030221_TE_SCE_dp_1 il a l'interrupteur donc je pense qu'il va pouvoir activer le mode en question

Furtif, 2023-07-27:
ou utiliser une rallonge découper les fils et un peut de scotch lol.

voir cette vidéo sa pourra vous éclairer.

Furtif, 2023-07-27:
> oui alors celui que je vous ai montré : https://www.amazon.fr/dp/B07MQ8GHG3/ref=pe_27091421_487030221_TE_SCE_dp_1 il a l'interrupteur donc je pense qu'il va pouvoir activer le mode en question

oui a l'interrupteur, voir... comme j'ai pas tester je sais pas. testez puis venais nous le dire.

just4545, 2023-07-27:
ça marche pas de soucis , je vous tiens au courant 
Merci pour tout :)

Furtif, 2023-07-28:
vais changer l'emplacement de cette conversation vais déplacer dans discutions car ceci n'est pas un issue logiciel mais un problème matériel.