cedricp/ddt4all #877: R link screen information missing after failed update from dealer

Nikoletapex, 2023-07-29:
Hi there..help please..anyone..i went my car to a renawlt dealer to upgrade system and maps..the dealer told me that the process got stuck two times and i left with problems at the screen..for example:it doesnt show information about the external temperature or the middle velocity and consumption as it used to display..the warning alarm when the car goes back ceased to work..the screen also doesnt show tyre pressure and others..its like he erased the potential to do any kind of measurements!apart from that the screen works, the speakers work ..i went the car back to the garage but he didnt kwow how to fix it..what do i do now?please please please help me im a woman and i dont know much of technology..

matzel82, 2023-07-29:
They have to give a lot attention to a OTS.... there is a patch to install after update. Don't make self thinks to your R Link !!!!!

matzel82, 2023-07-29:
I came from Germany and work in a Renault Garage too

AndyIreland, 2023-07-29:
is it a rlink1 or 2? they done 2 versions.

This is rlink1 in the picture , does yours look like this?:
[image]

AndyIreland, 2023-07-29:
> They have to give a lot attention to a OTS.... there is a patch to install after update. Don't make self thinks to your R Link !!!!!

what is an OTS ?

Nikoletapex, 2023-07-29:
> I came from Germany and work in a Renault Garage too

Whats an OTS?Im irrelevant with tecnology so please use as simple words as possible..thank you for your response and as you mention you are a technician at renawlt,so tell me why did this happen to my screen and how can i fix it..i read a lot of cases of damaged screens and i i can tell by common logic that my screen is still functionable..its like the dealer has erased some information or the commands with his machine..and the answear he gave me to get rid of me was..you are lucky your screen didnt get black..he didnt know how to fix it it was obvious..any ideas?

Nikoletapex, 2023-07-29:
> is it a rlink1 or 2? they done 2 versions.
> 
> This is rlink1 in the picture , does yours look like this?: [image]

Ive sent a picture of the screen.did you see it?thank you for your reply