cedricp/ddt4all #878: is it a rlink1 or 2? they done 2 versions.

Nikoletapex, 2023-07-29:
is it a rlink1 or 2? they done 2 versions.

This is rlink1 in the picture , does yours look like this?:
[image]

_Originally posted by @AndyIreland in https://github.com/cedricp/ddt4all/issues/877#issuecomment-1656707329_
         When the screen opens it sais rlink, plain  and its not coloufoul.the software system is 2.2.19.300
[image]

AndyIreland, 2023-07-29:
@Nikoletapex  - it looks like you have a rlink2 version there

Nikoletapex, 2023-07-29:
But when i open the screen it sais r link..anyway what should i do now to recover the computer back to normal?i think that the Renault dealer here has no idea of how to handle it..so your precious help is my only hope!
I forgort to mention one more problem..there was the potential to change the colour of the cadran(the plasce where it shows the speed of the car),blue,green,red and now it shows the menu complete but doesnt change the colour when you press the button..

AndyIreland, 2023-07-29:
> But when i open the screen it sais r link..anyway what should i do now to recover the computer back to normal?i think that the Renault dealer here has no idea of how to handle it..so your precious [...]

ah right - sorry i have rlink1so I know a bit about that. I am sure someone else on here will have idea like @KarelSvo and some others they are good on rlink 2 - IO think it has to be relaoded with mid and mid2 etc if there was an error but I am not well knowleged on rlink2 sorry, just some knowledge on rlink1 that is fitted to renault