cedricp/ddt4all #896: Lanr14 unlock

Serg797979, 2023-08-29:
Hello. help me remove the LANR14 anti-theft protection please.

rttr67, 2023-09-01:
I can help you with antitheft code if need
zaliavos.ud@gmail.com

Serg797979, 2023-09-01:
Hello. If you don't mind, please help. The radio that was on the LANR16 burned out. The new radio LANR14 is blocked. I have read dumps from both. Tell me, if I transfer it from my native radio to a new one, will it work?Получите Outlook для AndroidОт: rttr67 ***@***.***>Отправлено: пятница, сентября 1, 2023 2:07:35 PMКому: cedricp/ddt4all ***@***.***>Копия: Serg797979 ***@***.***>; Author ***@***.***>Тема: Re: [cedricp/ddt4all] Lanr14 unlock (Issue #896)
I can help you with antitheft code if need
***@***.***