cedricp/ddt4all #900: Problem audio megane 3 rs

Gaspaar13, 2023-09-10:
Hello, after being reprogram with clip
Something went wrong, don’t have values in acoustic and can’t write (wrong response) 
With Renault clio can’t reprogramar again because said that is already updated, this radio can be virginized? Or maybe save it with ddt?

thanks

shrlnm, 2023-09-10:
Hello @Gaspaar13 
what was reprogrammed ? LGU? MICOM?  from which version to which?

matzel82, 2023-09-10:
Do you read before/after steps ??
Are you a Renault Staff ?

Gaspaar13, 2023-09-10:
> from

Reprog audio, works fine (completed), but when I try again (looking to install version from a newer megane to get rs monitor) didn't do anything and said complete [image]

matzel82, 2023-09-10:
So...you need access to the Renault Server.... you need a USB Token.

Gaspaar13, 2023-09-10:
> So...you need access to the Renault Server.... you need a USB Token.

Only official usb or there is another way?

matzel82, 2023-09-10:
> > Do you read before/after steps ??
> > Are you a Renault Staff ?
> 
> Read yea 
> No, I'm not a Renault staff 

Normally it's prohibited to use the software. The Software calls at Home and transferred the VIN to Renault. The know where you are...

matzel82, 2023-09-10:
Especially... you can't get software from another car to yours....only with your vin connected "versions"

Gaspaar13, 2023-09-10:
Yes I think the same but worth the try [image]

Only need the missing script, it's the same radio and miu

Gaspaar13, 2023-09-10:
Maybe virginize?