cedricp/ddt4all #904: Rlink 2 screen rotation

Serg797979, 2023-09-17:
Hello. Radio LANR14 megane 4. After unlocking and configuring, I can’t turn the display image to landscape mode. How can I do it. And maybe the firmware version affects this? Firmware 2.2.17

AndyIreland, 2023-09-17:
could it be that this setting has to be changed here?:
[image]

Serg797979, 2023-09-17:
I tried. Does not help. After configuration it was set to landscape

matzel82, 2023-09-17:
Original Automatic configuration with clip

AndyIreland, 2023-09-17:
> I tried. Does not help. After configuration it was set to landscape

did you press hard reset after you made the changes or tried locking car up for 20 minutes to go to sleep and then come back to car to see if landscape happened after reboot of the rlink?

Serg797979, 2023-09-17:
> Оригинальная автоматическая конфигурация с зажимом
The radio is not from this car. And my clip doesn't find the configuration file for it.

Serg797979, 2023-09-17:
> > Я пытался. Не помогает. После настройки был установлен альбомный режим.
> 
> Вы нажимали полный сброс после внесения изменений или пытались запереть машину на 20 минут, чтобы заснуть, а затем вернулись в машину, чтобы посмотреть, произошел ли ландшафт после перезагрузки rlink?

I haven't tried it. I just pressed reset. Can hard reset help?

AndyIreland, 2023-09-17:
> I haven't tried it. I just pressed reset. Can hard reset help?

it does on rlink1 - it reboots the rlink. DO NOT whatever you do press 'SOFT RESET' !!

Serg797979, 2023-09-17:
> на моем rlink1 это происходит - он перезагружает rlink. **НЕ НАЖМАЙТЕ «SOFT RESET» !!**

Thank you. I will try.

shrlnm, 2023-09-19:
@Serg797979 
https://www.drive2.ru/b/643019689764615344/

Serg797979, 2023-09-19:
> @Serg797979 https://www.drive2.ru/b/643019689764615344/

Я этим способом и конфигурировал