cedricp/ddt4all #911: Renault Kadjar 2016 tried to update my software to v9 and now i have some problems. I managed to downgrade to v2 but i had some errors and now i have some issues.

Gorbanezu, 2023-09-24:


Gorbanezu, 2023-09-24:
[image]
[image]
[image]
[image]
[image]
[image]

matzel82, 2023-09-24:
@all I work at a Renault garage. Not every hardware is compatible to every software..

Gorbanezu, 2023-09-24:
> @ALL I work at a Renault garage. Not every hardware is compatible to every software..

So what can i do to repair my mistake?

Gorbanezu, 2023-09-24:
can i do something with virgin mode?

shrlnm, 2023-09-24:
@Gorbanezu https://www.drive2.ru/b/643019689764615344/

shrlnm, 2023-09-24:
@Gorbanezu How did you revert v2 from the v9? 
You are luky man!)
Try to use this instruction for recovering the configuration. https://www.drive2.ru/b/643019689764615344/

Gorbanezu, 2023-09-24:
> @Gorbanezu How did you revert v2 from the v9? You are luky man!) Try to use this instruction for recovering the configuration. https://www.drive2.ru/b/643019689764615344/

I downgraded to v7 and after that a modified version of 3.3 and after some other modified versions i managed to downgrade to 2.2 but now this problem

AndyIreland, 2023-09-24:
> [image] [image] [image] [image] [image] [image]

**It looks like just all those settings need inputting and your good to go**