cedricp/ddt4all #923: Acil

Ylmz038, 2023-10-17:
Hello, my vehicle is Renault Scenic 3. I do not know what I am doing while I am tampering with the ABS and ESP settings. The ABS light is constantly on. How can I turn it off and the hill start assist is not working, please help me.

Uploading VID_20231003_171717.mp4…

AndyIreland, 2023-10-17:
> Hello, my vehicle is Renault Scenic 3. I do not know what I am doing while I am tampering with the ABS and ESP settings. The ABS light is constantly on. How can I turn it off and the hill start assist is not working, please help me.
> 
> Uploading VID_20231003_171717.mp4…

hi, either enable ABS/ESP back again or extinguish the light by going into your 'clustre' settings and choose 'no ABS' and/or ' no eps'

Ylmz038, 2023-10-19:
Nasıl yapacağım bunu örnek bı fotoğraf var mi

Ylmz038, 2023-10-19:
Çok teşekkür ederim söylediğinizi deneyeceğim