cedricp/ddt4all #926: Hardcoded CAN  timeout for sending long commands.

shrlnm, 2023-10-21:
Hi. 
The CAN timeout is hardcoded in elm.py line 965. It do not alow to send long commands

I suggested to make a fix  https://github.com/cedricp/ddt4all/commit/629b672f5caff8c0e8df319ee050f3271e8132d6 
in my branch https://github.com/shrlnm/ddt4all

The idea is to use the timeout defined on the main screen.
But I have no chance now to test this fix now. Can some one test it and if it's ok I'll made a pull request.

Furtif, 2023-11-01:
merged thanks

closed 2023-11-01 by Furtif

Becraftshannon143, 2024-01-15:
@shrlnm need your help can u email me becraftshannon@yahoo.com about MEGANE 3 ac I saw a post u helped someone please my hubby is going crazy!!!

shrlnm, 2024-01-16:
@Becraftshannon143 let's discuss your question right here. It may help to improve this software.

Becraftshannon143, 2024-01-16:
Ok…went from manual to automatic air conditioner the heat and everything is
working but for the a/c the compressor is not kicking on.  I believe it is
the programming that needs tweaking.  If I give someone the vin can they
tell me exactly what to do….well tell my husband lol he’s been a zombie
over this for 2 weeks now!!!

shrlnm, 2024-01-17:
@Becraftshannon143
As I realise, your husband changed configuration of air conditioner and now a/c is working in unexpected ways and you are looking for a way to revert configuration in previous state. Am I right?  
The VIN and current dump may help in such cases.
If the configuration was not changed manually, then VIN will not help and you need to find the other cause of the malfunction

Becraftshannon143, 2024-01-17:
My husband is sleeping right now so I’m trying to get the info for you but
what is and how do I find the dump?  lol sorry if it’s something I can’t do
I will get back to you a little later it’s early am here

Becraftshannon143, 2024-01-17:
Sorry just woke up again and reread the message…not sure what change the
configuration means but if you mean changed the programming he did not do
anything with that nor did anyone that helped him.  So the A-C wasn’t
working fully afterwards and someone told him that after changing from
manual to auto u have to change the codes.  Is this correct?

Becraftshannon143, 2024-01-17:
VIN:  VF1BZAB0648137165
Still need to find out the current dump sorry so many messages….also just
wanted to point out that my husband is from Morocco and me New York….so
even though he learned English when we met things like this still get lost
in translation a bit too but pretty sure I have it right lol

Becraftshannon143, 2024-01-17:
> @Becraftshannon143 As I realise, your husband changed configuration of air conditioner and now a/c is working in unexpected ways and you are looking for a way to revert configuration in previous [...]

Just wanted to make sure I was replying to you bc for some reason it didn’t show your name when I was replying via email….

MehdiShanny, 2024-01-17:
> @Becraftshannon143 As I realise, your husband changed configuration of air conditioner and now a/c is working in unexpected ways and you are looking for a way to revert configuration in previous [...]

he is not sure what you mean by dump either...just waiting for that info and will find out.  once again lost in translation so sorry!!!

Becraftshannon143, 2024-01-17:
> For getting the dump file you need to run a pyren software under
> windows/linux/macos with option --dump. Scan the car, go to the appropriate
> ECU and select the "DDT : DDT screens". After running the new ddt window
> the dump file will be created in folder pyren3/dumps

Do you know a tech who will do this online if I give them access somehow?
Of course will pay.  If not I know my husband can do it.  I think he is
just getting worried now

MehdiShanny, 2024-01-18:
Ok thanks a bunch we are working on getting the software ourselves and everything thanks again!  Will be in touch soon!