cedricp/ddt4all #931: ddt4all crashes when trying to load screens

neilnutt, 2023-11-01:
I am able to load ddt4all and complete a scan. ddt4all unexpectedly quits a short moment after I click through to see a screen (screen never shows).  It doesn't matter which ecu I try to access, they all get the same unexpected quit with no warning raised.

Screenshot showing last moment before crash
[image]

Log file
[elm_ddt.log](https://github.com/cedricp/ddt4all/files/13232192/elm_ddt.log)

AndyIreland, 2023-11-01:
what happens if you give this version a try - choose 'all cars' it has a built in database 
[https://www.drive2.ru/b/498093336985338243/](url)

neilnutt, 2023-11-02:
Yes. The copy on the Russian site works. Doesn't resolve the issue for the current release, but fixes it for me.

Furtif, 2023-11-03:
🧐

KarelSvo, 2023-11-07:
> I am able to load ddt4all and complete a scan. ddt4all unexpectedly quits a short moment after I click through to see a screen (screen never shows). 

https://github.com/cedricp/ddt4all/discussions/930#discussioncomment-7490140