cedricp/ddt4all #932: DDT4ALL plante lors de validation calculateur - win 7 32 bits

Ludo25081985, 2023-11-05:
Bonjour,
Tous d'abord merci à  Cedric pour sont logiciel.
J'ai un souci lors de mon installation, je suis en windows 7 32 bits.
J'installe la version Python 3.8.6en x86, j'installe la base de données, tous ce passe bien, le logiciel trouve 2138 calculateur au démarrage.
J'ai fait un essai en mode non connecté, je valide véhicule,  TBD, le calculateur voulu , et des que je reclique dessus pour afficher les possibilités le logiciel plante sans que j'ai le temp de voir ce qui s'affiche sur l'autre page de commande.
J'ai essayé d'autre version , il y'a que celle ci qui marche, sur n'importe quelle autres calculateur sa fait pareil.
Qu'elle serais la cause de se plantage s'il vous plaît ?
Si un âme charitable pourrais m'aider ce serais sympa merci.

AndyIreland, 2023-11-05:
Hello - try have a go and try and download this one and see if it crashes , its got the database built in , choose "All Cars" 
https://www.drive2.ru/b/498093336985338243/
- uninstall the one you have installed first.

Ludo25081985, 2023-11-05:
Est ce que ça prend en compte le 32 bits , je ne pense pas si ??

Ludo25081985, 2023-11-05:
Merci pour le retour déjà c'est cool

Ludo25081985, 2023-11-05:
Je vais essayer quand même,  mais j'aurais préféré utilisé la version de cedric, je pense que c'est juste une modification à faire sur fichier écu.py mais je ne sais pas comment l'ouvrir ni l'écrire.

Ludo25081985, 2023-11-05:
Ça marche avec le lien russe merci Andy,  mais si quelqu'un à la solution pour le logiciel originel de cedric je serais preneur.