cedricp/ddt4all #934: Downloaded ddtall-win-installer-23.06.13-python-3.8.6-32bits.exe but opens up version 2.1.1

AndyIreland, 2023-11-08:
I downloaded version 3.8.6 from the release area - I chose to place a link on desktop to start, I click on desktop icon to start and it starts up version 2.1.1 of DDT4ALL what have I done wrong?
I am working with win32 - i dont want to use the 64bit version 

[image]

Thanks

KarelSvo, 2023-11-08:
3.8.6 python version
2.1.1 ddt4all version

AndyIreland, 2023-11-08:
This is what target my desktop icon opens up:
"C:\Program Files (x86)\ddt4all\Python38-32\python.exe" "C:\Program Files (x86)\ddt4all\main.py"

KarelSvo, 2023-11-08:
And where is the problem now?

AndyIreland, 2023-11-08:
> And where is the problem now? C:\Program Files(x86)\ddtall\Python38-32/NEWS

I click on the desktop icon and it starts version 2.1.1 :
[image]

KarelSvo, 2023-11-08:
Yes, that is correct, it is determined by the version.py file https://github.com/cedricp/ddt4all/blob/v2.1.1/version.py
And now the problem?

AndyIreland, 2023-11-08:
I dont get it. If I have installed ddtall-win-installer-23.06.13-python-3.8.6-32bits.exe ... and it places a desktop shortcut when installed , and I press on the icon on the desktop why is it opening up the version 2.1.1 DDT4ALL and not opening up a version of 3.8.6 DDT4ALL?
I cannot see how I can explain it any more clearer

KarelSvo, 2023-11-08:
Because 3.8.6 is the version of the Python programming language and not the version of the ddt4all software.

ddtall-win-installer  **software name**
-23.06.13 **date** 
-python-3.8.6- **python version**
32bits.exe  **windows version**

AlMeu, 2023-11-08:
@AndyIreland you give advices at many here, but you don't make difference between pyton version and ddt4all version?

KarelSvo, 2023-11-08:
[This](https://github.com/cedricp/ddt4all/discussions/930#discussioncomment-7490140)   is no longer a problem because no reaction from you?

AndyIreland, 2023-11-08:
> 3.8.6 is the latest possible version for Windows 7. https://www.python.org/downloads/windows/

ah right that makes sense now - I thought maybe there was a newer version of DDT4ALL called version 3.8.6 
sorry I was confused

Furtif, 2023-11-11:
https://github.com/cedricp/ddt4all/discussions/933#discussioncomment-7541397