cedricp/ddt4all #935: rying to change my radio

mihirat33, 2023-11-08:
Trying to change my radio on Renault scenic 3, when I attempting to connect with audio ecu I receiving  a msg :
Found ECU [Audio] (no relevant ECU file found) :DIAGVERSION [164] SUPPLIER [BT2] SOFT [1003] VERSION [0000]
Is a DB issue and how to solve.

AndyIreland, 2023-11-08:
have you re-wired your ELM327 or Rewired a OBD2 fly lead? 

you need to wire it up like this so you can read multimedia through the OBD2 socket
https://www.gps-rlink.com/r-link-1/activate-android-auto-waze-on-r-link-1/

mihirat33, 2023-11-08:
This one 
[image]
trying to change with this
[image]

KarelSvo, 2023-11-08:
R1_08_V1.3_20090612T085711.xml

AndyIreland, 2023-11-08:
if you want the xml file I have uploaded it to here for you. - I wonder if you will have to fill out the VIN details as well if its come out of another car? 
https://we.tl/t-8QkL0TbcgP

KarelSvo, 2023-11-08:
The file is not missing, it just needs to be selected manually because the Autoident data is incomplete.