cedricp/ddt4all #977: Dil destegi

bersa2009, 2023-12-28:
Program kurukurken Türkçe dil desteği yoktur 😔 lütfen türkçe dil destegi bırak

bersa2009, 2023-12-28:
[Bersa2009](https://github.com/bersa2009) yorum yaptı [şimdi](https://github.com/cedricp/ddt4all/issues/977#issue-2058718872)
Program kurukurken Türkçe dil desteği yoktur 😔 lütfen türkçe dil destegi bırakın

Furtif, 2023-12-29:
Support is in https://github.com/cedricp/ddt4all/tree/master/dtt4all_data/locale/tr/LC_MESSAGES

bersa2009, 2023-12-29:
While installing the program, there is no Turkish language in the exe file. When you leave it later, Turkish language cannot be translated... When you download the Turkish package from githup.com, you leave it in the c// folder. When you open the program, there is no Turkish language... When you download the source files from githup.com, there is a Turkish file. These source files are exe It would be easier if there was a translated file, if there was a Turkish language test when installing… I cannot change the language because there is no other explanation.

bersa2009, 2023-12-29:
Could you please send me the exe file via e-mail? 🙏 Turkish language support..

bersa2009, 2023-12-29:
When loading the exe file, it shows 6-7 or maybe more languages ​​and there is no Turkish in it. Can you send me a ready exe file in Turkish for Windows 10 64 bit computer?

Furtif, 2023-12-29:
uses https://poedit.net/