cedricp/ddt4all #1013: Renault Megane 4 Automatic High Beam and Lane Assist Fault  "Fault Code: No Camera Multiple Signal"

engilis, 2024-01-22:
Hello, I am the owner of a Megane 4 Sedan 2016 model vehicle. My vehicle's battery was weak and a message saying **"Check Automatic High Beam"** appeared on the screen. I thought that this message would disappear when I changed the battery, so I ordered a new battery, removed the old battery and installed the new battery while the engine was not running. I did not use any 12 V DC voltage source during the change. I took it apart and replaced it. The signal still continues. I deleted the fault codes from the mobile phone with the Car Scanner application using ELM 327, but it did not fix it. I installed the PyRen program on my laptop and connected with it. There were 10-15 error codes in many different places. I deleted them and the remaining 2 error codes remained. There is 1 fault code left on the Instrument Panel and 2 on the UCH. Error Codes: DTCC18287 (DF494) and DTC95D887 (DF 389) **NO CAMERA MULTIPLEX SIGNAL**. These are also malfunctions of Automatic High Beam and Lane Keeping Assist. These faults were not cleared. I couldn't reach a conclusion with Car Scanner and PyRen. Will I change the configuration with the DDT4ALL program? How will I do? Or will the authorized service intervene and fix it with the Reno Link program? I would be happy if you help. Thanks in advance

matzel82, 2024-01-22:
It's possible to get a automatic configuration after change. Better to go Renault Garage....

I don't know this problem. To try this with Pyren or other... I don't know 🤷🏼‍♂️ it's risky.

AndyIreland, 2024-01-22:
Sometimes with PyREN you need to clear DTC error codes with conditions.
The conditions could be that:
You have to clear the errors with engine on,
or you have to clear errors with the engine off,
or you have to have the car in 'park mode' (or ACC mode) thats with the keycard in slot and long press on the start button. 

if you get a 'RSP57' response (I think it is in PyREN) when clearing codes I think it is when clearing the codes I think you are OK but if its another number then 'conditions were not met' and the code stays and is not removed

KarelSvo, 2024-01-22:
https://github.com/cedricp/ddt4all/discussions/1000#discussioncomment-8124844