cedricp/ddt4all #1014: I cant't startup this software.

Eksponat, 2024-01-23:
Help please, i dont know how startup this software. 
[image]

AndyIreland, 2024-01-23:
haven't you got an old windows laptop you could try using instead?

Eksponat, 2024-01-24:
unfortunately no

AndyIreland, 2024-01-24:
https://github.com/cedricp/ddt4all/issues/122

Eksponat, 2024-01-24:
Left code: my fix
Right code: original
If anyone has the same problem, here is the solution code:
        langlayout = widgets.QHBoxLayout()
        self.langcombo = widgets.QComboBox()
        langlabels = widgets.QLabel(_("Interface language (need save and close)"))
        langlayout.addWidget(langlabels)
        langlayout.addWidget(self.langcombo)

        if 'LANG' in os.environ:
            for s in options.lang_list:
                self.langcombo.addItem(s)
                if options.lang_list[s].split("_")[0] ==os.environ['LANG'].split("_")[0]:
                    self.langcombo.setCurrentText(s)
        else:
        # Handle the situation when "LANG" is not present in the environment
            print ("The 'LANG' variable is not set in the support environment.")

        layout.addLayout(langlayout)

[image]

Furtif, 2024-01-24:
Install requirements 🤦🏻‍♂️

Eksponat, 2024-01-24:
I did everything that was required, including in #122, none of the suggestions corrected the situation.

Furtif, 2024-01-24:
> I did everything that was required, including in #122, none of the suggestions corrected the situation.

correct situation is: 
``` 
if not os.environ["LANG"]:
  export "LANG" = "en-US" 
```

more or less this.

or try :  ... except: export "LANG" = "en-US"

Furtif, 2024-01-24:
ex: 
```python
       # languages setting
        lgn = None
        try:
            os.environ['LANG'].split("_")[0]
        except:
            lgn = "en"
            
        langlayout = widgets.QHBoxLayout()
        self.langcombo = widgets.QComboBox()
        langlabels = widgets.QLabel(_("Interface language (need save and close)"))
        langlayout.addWidget(langlabels)
        langlayout.addWidget(self.langcombo)
        for s in options.lang_list:
            self.langcombo.addItem(s)
            if options.lang_list[s].split("_")[0] == lgn:
                self.langcombo.setCurrentText(s)
        # self.langcombo.setCurrentIndex(0)
        layout.addLayout(langlayout)
        #
```

this teoricaly set English by default is not exists

Furtif, 2024-01-24:
or set environement var 

```
os.environ['LANG'] = "en_US"

```

Furtif, 2024-01-24:
if ok close this issue. 
Thanks.

Furtif, 2024-02-08:
pip3 install -r requirements.txt

Le-Spirit, 2024-02-09:
^ merci / thank you Furtif, previous post updated and enhanced.

tLupus1978, 2024-02-11:
great, thanks for this perfect explanation!

commit aafe90bc (2024-01-24): try fix #1014
closed 2024-03-05 by Furtif