cedricp/ddt4all #1017: problems with installing ddt4all on linux

Kerel007, 2024-01-27:
Hi
I have a problem installing ddt4all on Linux.
After entering the command:

$python3 main.py

It gives an error:

Traceback (most recent call last):
  File "/home/XYZ/Games/ddt4all-3.0.0/main.py", line 15, in <module>
    import PyQt5.QtWebEngineWidgets as webkitwidgets
ImportError: /home/XYZ/.local/lib/python3.11/site-packages/PyQt5/Qt5/lib/libQt5WebEngineCore.so.5: undefined symbol: _ZNSt12out_of_rangeC1EPKc, version Qt_5

Thank you in advance.

KarelSvo, 2024-01-27:
lsb_release -a
python  --version
python3 --version
dpkg -l | grep qt

Furtif, 2024-01-31:
$python3 -m pip -r requirements.txt

Furtif, 2024-01-31:
python 3.12 for better

Furtif, 2024-02-03:
close issue if ok. 
Thanks