cedricp/ddt4all #1018: Bad ELM response: WRONG RESPONSE : NR: Request Out Of Range(7F2231)

x5nder, 2024-01-28:
Before I'm going to try and convince a Renault dealership to unlock the CGW on my Clio V with CLIP, I wanted to do a few test runs so I know where to find all parameters that I want to change in ddt4all. So when I connect, select the correct car (CLIO V), select one of the ecus (A-IVI v1.10) and then the correct screen (Navigation > System Information $DAB1), I get actual read-out values in green, but the bottom log window of ddt4all is giving me tons of "Bad ELM response: WRONG RESPONSE : NR: Request Out Of Range(7F2231)" errors. Is this normal, or am I doing something wrong? I am assuming the read-out values in green mean that the ELM327 scanner is working properly and that the errors either mean specific parts of the ecu can't be read, or it's because the CGW is not unlocked yet... am I mistaken?

AndyIreland, 2024-01-28:
could be the gateway ..... but why are you, or it sounds like you are , selecting the ECU's manually , Why dont you try searching for the ECU's in your car with the magnifying glass in DDT4ALL?

x5nder, 2024-01-28:
Hmm. You might be onto something here. If I do a scan, I'm getting (in green) the following ecus:

[VSP] VSP_v1a3
[MIU] Central Panel CMFB/CDp2_7/9.3_2
[USM[ USM_C1A_SW6
[HFM] C1A_HFM_DDT_7_1
and one or two more (can't remember)...
...but none of the [Navigation] ecus. 

The last Navigation ecu 'version' for my Clio is A-IVI v1.12... does this mean I should find a newer version somewhere, or should I draw a different conclusion? The ecus in green can be read fine according to the log ('ECU uses SDS 10xx' and 'Switching to session mode 10xx') but when I try to access the Navigation ecu, that's when I'm getting the error mentioned in the title of this topic.

x5nder, 2024-01-28:
Hmmm. I found another ecu update where the latest Navigation ecu version is v2.8, but even here none of the navigation ecus are recognized. I know that my EasyLink head unit was very recently updated to the latest version; could it be that there is no ecu.zip file yet that has the correct Navigation ecu version for the current EasyLink software version installed in my car?

AndyIreland, 2024-01-28:
yeah 2.8 is the latest I can see on mine too:
[image]

I wonder if for theClio5 you still have to make up a special splitter cable or solder the ELM327 to read CAN2 / CAN HI like you did with the earlier radionav's or whether thats eliminated on the Clio 5 now? 

When I have seen people 'apparently' bypassing the security gateway that have took apart the dash and wired the ELM327 directly to the back of the BCM or nav wiring to change parameters on the radionav

x5nder, 2024-01-29:
I found my old screenshots, and in those I can see that I actually had access to A-IVI v1.10 without errors until 2 weeks ago. I am going to assume that the software update performed by the Renault dealer last week (to v2.13 / 04-2023) updated the Navigation ecu to a version that's not (yet) supported by the existing ecu databases.

Now the main question is: is there a way to obtain updated ecu files? Where do they come from? Are they extracted from the official CLIP software? I am assuming all current Clio VI cars (ie, the 2024 model) will run on the same updated ecu version, so at some point there has to be support for them.

Maybe this is even a question for @cedricp as he's most intimately knowledgeable on this topic.

AndyIreland, 2024-01-29:
OK, so I cannot find this article I read (typical!) and i have been in my history and bookmarks. There was a post somewhere and I am pretty sure it was for the easylink , that said something like: 
"If you are wanting to enable Wireless Android Auto & Apple Car Play be sure to do it before update so and so" because from what I made out of that is Renault has updated the software/firmware of the easylink to close the loop hole of people being able to change the settings AFTER this update!! - so it could be that. 

Also Cedric P program (DDT4ALL) does not even touch the ECU Database, in fact originally you had to download DDT4ALL originally and source your ecu.zip file from other sources (namely latest DDT2000 Database) so i dont think that will help. - I cannot see you even getting a more up to date database as well , possibly you might but I think the problem is not the actual database , but the fact they done the software update on your easylink unit to v2.13 and that has closed up that loophole that was there on the version before it.

x5nder, 2024-01-29:
Ahhhh, damnit... I never read that piece of information, otherwise I would have made sure to take care of this before the update. Ah well... nothing that can be done about that (pretty sure it's not going to be easy or even possible at all to roll back to a previous version). Thanks for the info, though...!!!!

x5nder, 2024-01-29:
No worries-- it definitely looks like Renault disabled access to the Navigation ecu in their 2023 updates. I should have been smart enough to get the CGW unlocked first to make the changes I wanted, before letting Renault update my system.

I was mainly looking into:
* enabling wireless Carplay (even though I'm currently using a wireless dongle that works quite well);
* changing L/100km to km/L;
* enabling MEX (multi-sense) so I can enable Eco mode as default;
* change the startup logo.

But all of these are done on the Navigation parameter page, so... tough luck. 

I might try to install CLIP itself (found a cracked version from September 2023) to see if that will offer me any opportunities, but I seriously doubt it.

KarelSvo, 2024-01-29:
[here](https://www.clio5-forum.de/forum/thread/976-ddt4all-gateway-router-mod/) and[ here](https://www.clio5-forum.de/forum/thread/955-ddt4all-clio-v-mod/)