cedricp/ddt4all #1022: ECU: Busy Repeat Request(7F3E21)

danir-de, 2024-02-06:
Hello, I'm having a problem with my radio configuration, where the `[ RADIO MID DAB ] Radio 2012 DAB v4.0` screens have become unresponsive and all speaker and radio (AM/FM/Digital) are unusable and locked up on my RLink Evolution.

interface got unresponsive, after I tried setting a dummy value at the `1 - Configuration` > `15 - Radio reception` dialogue.
(I had to enter something, because the value was lost and the radio had the orange exclamation mark "Radio unconfigured") after saving an unrelated setting under `12 - ECU configuratio`.

Now the DDT4ALL interface just returns
```log
Bad ELM response : WRONG RESPONSE : NR: Busy Repeat Request(7F3E21)
``` 
and looks interface looks like this:
[image]
[image]

and I can't seem to be able to read any settings from the `[ RADIO MID DAB ] Radio 2012 DAB v4.0` screens.

This is also troublesome, since the controller device thats locked up doesn't seem to go to sleep together, when the car is locked and it already emptied my car battery.

All the menues worked before setting the `15 - Radio reception` values and my car isn't under warranty anymore. 
The R-Link Evolution firmware versions are application 11.347.3272159-8063 and GPS 2.18.911.318008.
Any buttons in the radio menu are greyed out, I can't even change between AM, FM or digital or change bands, since all corresponding buttons are greyed out or unresponsive.

Any help is appreciated! Kind regards.

AndyIreland, 2024-02-06:
hello there - have you got the converted  y-cable (or soldered ELM327) to read CAN2/CAN Hi? 

What ELM327? - wi-fi, bluetooth or USB?

AndyIreland, 2024-02-06:
my backup just has a load of 00000000000's in the boxes

[image]

danir-de, 2024-02-06:
I use [this ELM327 USB Adapter](https://amzn.to/3uw0mqj).

I've modified the cable like this:
[image]

danir-de, 2024-02-06:
> my backup just has a load of 00000000000's in the boxes [...]

How many 0s? Could you please paste it here?

AndyIreland, 2024-02-06:
0x00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

AndyIreland, 2024-02-06:
havent got anything written in the digital radio boxes because my car (Renault Zoe EV) has not got DAB - just FM

danir-de, 2024-02-06:
I've tried to overwrite the value, but I'm still getting 
```code
Bad ELM response : WRONG RESPONSE : NR: Busy Repeat Request(7F3E21)
```

I suppose there is no direct way to restart the device, when a hard reset doesn't work?

[image]

AndyIreland, 2024-02-06:
> I use [this ELM327 USB Adapter](https://amzn.to/3uw0mqj).
>
> I've modified the cable like this: [...]

yeah thats right  - try increasing the CAN Timeout on DDT4ALL to 600ms 
what car have you?

AndyIreland, 2024-02-06:
> I've tried to overwrite the value, but I'm still getting
> 
> ```
> Bad ELM response : WRONG RESPONSE : NR: Busy Repeat Request(7F3E21)
> ```
> 
> I suppose there is no direct way to restart the device, when a hard reset doesn't work?
> 
> ![image](https://private-user-images.githubusercontent.com/4672840/302786002-a1b2855a-1de8-4ed4-8927-e962093401cd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcyNTYyOTYsIm5iZiI6MTcwNzI1NTk5NiwicGF0aCI6Ii80NjcyODQwLzMwMjc4NjAwMi1hMWIyODU1YS0xZGU4LTRlZDQtODkyNy1lOTYyMDkzNDAxY2QucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDZUMjE0NjM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjhkZGFmMDY4MjI3NjVlOGVjYjk3ZDA5NDJhMGQ2NDM4ZGEwMThjNmVhMGNkY2ViYzg1ZmIyMGYwNTcyMWNmMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.4Y6cDWngHUg0ejmb28_yfRiCYeVD5aAy3REGFk4nTF8)

Try pressing the home button on the dashboard 5 times to reboot the rlink

danir-de, 2024-02-06:
The RLink menu `MFD v5.4` is and working before. 
Restarting it by pressing the home button several times sadly didn't change the behavior, both from the radio unit, the speakers or the error when trying to use the `[ RADIO MID DAB ] Radio 2012 DAB v4.0` menus.

AndyIreland, 2024-02-06:
Yeah - the MFD goes hand in hand with RADIO 2012 - have you made sttings to the MFD under Navigation area?

danir-de, 2024-02-06:
> yeah thats right - try increasing the CAN Timeout on DDT4ALL to 600ms what car have you?

Other then the error message code changing to 
```log
Bad ELM response : WRONG RESPONSE : NR: Busy Repeat Request(7F3E21)
```
nothing happens. 

I have a Renault Clio IV (Facelift) from 2018.

danir-de, 2024-02-06:
> Yeah - the MFD goes hand in hand with RADIO 2012 - have you made sttings to the MFD under Navigation area?

Yes I did.

AndyIreland, 2024-02-06:
the reason I asked what car you have is because I think on Petrol & Diesel cars when you have that message of unconfigured device etc is because petrol & Diesel have a BIC Ecu that has to be properly configured too

AndyIreland, 2024-02-06:
have you got a PIC18F25K80 chip in your USB ELM327?

danir-de, 2024-02-06:
The unconfigured device message went away, when I saved the empty radio `15 - Radio reception` parameter.
But at the same time, audio, radio and the `[ RADIO MID DAB ] Radio 2012 DAB v4.0` menu started failing.

danir-de, 2024-02-06:
> have you got a PIC18F25K80 chip in your USB ELM327?

I'm unsure on how to really find that out. It self reports as 0x7523:0x1a86.
By digging deeper, it seems to have a [CH340](http://www.wch-ic.com/products/CH340.html) chipset of some sorts.

AndyIreland, 2024-02-06:
> > > I use [this ELM327 USB Adapter](https://amzn.to/3uw0mqj).

Them forscan ones are not wired up normally for reading CAN HI in Renault - and I doubt its got the correct PIC chip in it ... maybe this why you got bad elm response

AndyIreland, 2024-02-06:
> > have you got a PIC18F25K80 chip in your USB ELM327?
> 
> I'm unsure on how to really find that out. It self reports as 0x7523:0x1a86. By digging deeper, it seems to have a [CH340](http://www.wch-ic.com/products/CH340.html) chipset of some sorts.

Take it apart physically and with a magnifying glass look whats written on top of the chips on the main board - unless it said in the listing if it had the PIC chip which I dont think they do - if you have trouble reading and writing the next ELM327 you should look for on amazon when you are doing a search is for PIC18F25K80

danir-de, 2024-02-06:
Do you think this one here looks right? 
https://aliexpress.com/item/1005004078281314.html

I'd like the next one to be USB also, and they are pretty hard to find.

AndyIreland, 2024-02-06:
> Do you think this one here looks right? https://aliexpress.com/item/1005004078281314.html
> 
> I'd like the next one to be USB also, and they are pretty hard to find.

seems to tick the boxes (as long as the listings are proper and they send you that one) - I have seen people put the PIC chip on their listings and when someone ordered it was completely different from listings or they have split it open and there is no required PIC chip even in there! - if there is a way to rip some people off these days and get away with it they will some people! 

Dont forget whatever one you buy you will have to re-wire it to read and write CAN HI/CAN2 over OBD2

danir-de, 2024-02-06:
[Isn't this enough](https://github.com/cedricp/ddt4all/issues/1022#issuecomment-1930782405)? I've re-wired a connector cable, isn't it enough continue using that?
I've now ordered the one from above from AliExpress and [this one from Amazon](https://amzn.to/3UykPFS), just to have a backup if one doesn't work.

Thanks for the help, its very much appreciated!

AndyIreland, 2024-02-06:
> [Isn't this enough](https://github.com/cedricp/ddt4all/issues/1022#issuecomment-1930782405)? I've re-wired a connector cable, isn't it enough continue using that? I've now ordered the one from above [...]

yes if you have wire up the cable as per diagram it should be able to read and modify rlink through obd2 socket

good luck.

KarelSvo, 2024-02-07:
https://github.com/cedricp/ddt4all/discussions/967