cedricp/ddt4all #1038: R-Link 2 24 pin diagram

larskka, 2024-02-26:
Hy hope someone can help me. have a scenic iv from 2020 where i want to install reverse camera. anyone can help with a diagram to r-link 2 with 24 pin have tryed 18 cam+ 19 cam - 20 cam + suply but i dont work. thr camera an wire work when i connect to another schreen, so i think it is the wiring there is wrong...

larskka, 2024-02-26:
[image]
this is what i have tried drawn as looking inside the connection on r-link

AndyIreland, 2024-02-26:
if its a car without a reverse camera and you have put it on as an add on/upgrade I should imagine you will have to enable rear reverse camera in the navigation section with DDT4ALL on your rlink2

larskka, 2024-02-26:
Yes its is new and activated with ddt4all. it has now the camera menu, but says defet camera or missing signal. so i think i dident use the right pins? i have bought at adapter  but it is 32 pin and the radio has 24 pin so i found the connections on the picture but i dont work

larskka, 2024-02-26:
it is 6 volt. if i connect camera wit power from extern cable there is still no video signal on r link . but there is picture if i connect the video cable to a old tv...

AndyIreland, 2024-02-26:
strange , should work then you would think. - so you are using the +6v and -6v pins and the CVBs  + and - and putting car into reverse gear and no pic ? - where did you get the 24pin wiring detail diagram from. 

here is wiring for rlink1 but I dont know if it the same for rlink2:
[image]

[image]
[image]
[image]

larskka, 2024-02-26:
I think I saw it on this site and it fit fine with the pins that were not in use. so just like you have shown here but - for the camera supply I have taken out the rear lights. But it is not only the power that has problems, because I have tried to supply the camera with external power, so the video signal is also wrong. When the car is put into reverse gear, a black screen appears and it reports a camera error or lack of signal

AndyIreland, 2024-02-26:
yes, it could possibly be different wiring for the rlink2 vs  the rlink1 24 pin connector then

larskka, 2024-02-26:
My thought is that mabye there i a different wirring in r-link2 ?

AndyIreland, 2024-02-26:
I found this - pin 21 could it be on the 32pin cable?
https://github.com/cedricp/ddt4all/issues/696

[image]

larskka, 2024-02-27:
Yes it would be great, but im not sure if the wirring is correct. Its for the elder medinav, i have seen som sellers write , that it wont support r-link?

matzel82, 2024-02-27:
You need a diagram for the connection ? I can give you… I need your VIN

the lead at the picture is for MediaNav UlC4 not for RLink. It’s different.

matzel82, 2024-02-27:
I think that’s important to config the camera at RLink. To „yes“

larskka, 2024-02-27:
Vin nr.  vf1rfa00063460938 thankyou

larskka, 2024-02-27:
> You need a diagram for the connection ? I can give you… I need your VIN
> 
> the lead at the picture is for MediaNav UlC4 not for RLink. It’s different.

vf1rfa00063460938 thank you :-)

matzel82, 2024-02-27:
Apropos… you have to visit a garage, there is a OTS to reprogramming your ECUs

matzel82, 2024-02-27:
Im not shure if you have the correct connection. Normally you use the connection from „Radio“ behind the Clovebox. There I can’t find so long the diagram. But I have attached you the manual to install the camera system originally. There are the correct instructions an a diagram to connect. I hope it’s helpful.
[EBA8201672189.PDF](https://github.com/cedricp/ddt4all/files/14418671/EBA8201672189.PDF)

larskka, 2024-02-27:
Thank you so much :-) my fault i have trying to connect on the back of  r-link screen an not behandling the clovebox. So i learned a lesseon today

AndyIreland, 2024-02-27:
so did you have to join your camera wires to the 32pin connector then?

larskka, 2024-02-27:
[image]
Now i found a 32 pin over clovebox.  But there is cable in the pins i think i should use Brown, yellow, red and black . Is it the right cable? If shall i just cut the cable and add the ones from camera? And finale page 22 in the instruktion  what Are the cables A? Are you lookind in the plug from the behind where you plug in the cable?  Is 21 video + 22 video -  23 supply + 24 supply - ? Thank you for your time:-)