cedricp/ddt4all #1056: Cofiguration for new rlink 2 rlan16

sdade86, 2024-03-23:
Hello all gentlemen!
I decided to mount a used r-link2 system, version lanr16 281152288R hw: 3.11.5.4 sw: 7.8.24.136 instead of the original lanr14 on my Talisman Initiale 2016.
Once installed I immediately noticed that I was quite lucky as it turned on without any problems with the anti-theft device.
But now there are problems. the joypad in the central console doesn't work at all while of the steering wheel controls only the button to lower the volume and change the source works.
I have already done several tests with ddt4all but I can't get it to work correctly.
Furthermore, I cannot find the correct xml file as the program always tells me that the control unit software does not match perfectly.
Please let me know if you need any other information and how to get it.
Thank you all!

sdade86, 2024-03-23:
Just au update, joypad workin after can clip parameter update

sdade86, 2024-03-23:
another proble is the hu display not showed in rlink2 setting

sdade86, 2024-03-23:
I'm sorry, I just realized I wrote in the wrong section

shrlnm, 2024-03-24:
@sdade86 https://www.drive2.ru/b/643019689764615344/

sdade86, 2024-03-24:
Thanks a lot for the link! It is very helpfull! Put the ecu in Virgin mode is mandatory for use macro? Or tha macro can config anyway? Other question is that: the radnav actually have different vin from car. I need to change it for macro? (If yes maybe i have answer for first question)

sdade86, 2024-03-24:
does the macro also restore the original audio parameters?

sdade86, 2024-03-24:
> @sdade86 https://www.drive2.ru/b/643019689764615344/

Thanks a lot for the link! It is very helpfull! Put the ecu in Virgin mode is mandatory for use macro? Or tha macro can config anyway? Other question is that: the radnav actually have different vin from car. I need to change it for macro? (If yes maybe i have answer for first question).
does the macro also restore the original audio parameters?

sdade86, 2024-03-24:
> @sdade86 https://www.drive2.ru/b/643019689764615344/

@shrlnm 

Thanks a lot for the link! It is very helpfull! Put the ecu in Virgin mode is mandatory for use macro? Or tha macro can config anyway? Other question is that: the radnav actually have different vin from car. I need to change it for macro? (If yes maybe i have answer for first question).
does the macro also restore the original audio parameters?

sdade86, 2024-03-25:
Hello, today i have try the bot but not give me any file. Bot sayd data is ready and stop

shrlnm, 2024-03-25:
@sdade86 You need to generate macro for your vin with your current 76: option. 
Virgin mode require in some cases when handsfree not working for example. 
Try to start without virgin. If something will not work then try with virgin mode.

I just checked the bot and it's working. Try once again. If you will get no success then share your vin

sdade86, 2024-03-25:
I have try again now but norhing. 
this is the command /vin VFR1RFD00854631851 76:281152288R
for vin and ecu ref
[image]

sdade86, 2024-03-25:
I just realized I made a mistake when I wrote the wine... sorry, I'm really stupid!

sdade86, 2024-03-25:
Ok, i have run macro created from the bot and hud come back! thanks a lot! but swc have same problem, you think need to put in virgin mode?

sdade86, 2024-03-28:
I'm back! yesterday I took a lot of courage and pressed the virgin button in dev mode. then I reconfigured with the bot macro and everything was perfect again (even the language in the notifications which before virgin mode did not match that choice)... everything except SWC. At this point I became convinced that it had to be a HW problem. So I re-unmounted rlink2 from the machine. I opened it and found the damage fortunately repairable with some soldering skills and some smd capacitors!

@shrlnm I really appreciate your work and your help, I would like to contribute with a donation, do you have paypal?  A thousand thanks