cedricp/ddt4all #1064: ddt4all obdII elm327 Bluetooth com 3/4 access denied

letrasluis2, 2024-03-31:
Hi, 

I cannot use DDT4ALL version 2/3/5 with Win11x64 and ELM327 Bluetooth - permission error for COM3 and COM4.

I have used different ddt4all versions in order to solve connectivity but there is no joy.

I have reset pc, and Bluetooth devices.

I have seen this errors in some Internet posts but than the discussion summary is lost so I have no idea how can resolve
[image]

[image]
[image]

[image]

KarelSvo, 2024-04-01:
email?

letrasluis2, 2024-04-01:
> email?

letrasluis2@gmail.com

KarelSvo, 2024-04-01:
OK delete email

KarelSvo, 2024-04-01:
email sent

letrasluis2, 2024-04-02:
> email sent
meanwhile I arrange usb device, and I have tested it with your sw: ELM not responding

[image]
[image]
[image]
[image]

letrasluis2, 2024-04-02:
> email sent

with elm Bluetooth

[image]

COM 4 overflow

[image]

letrasluis2, 2024-04-02:
> email sent

and whith old elm Bluetooth 
[image]

com 3 show some results and com 4 overflow in connection test...

[image]
[image]
[image]

KarelSvo, 2024-04-03:
The recommendation is, to replace the unsuitable adapters with a vLinker FS USB. v1.1 ATPPS [FAIL] shows, it is a very bad copy.

letrasluis2, 2024-04-03:
> The recommendation is, to replace the unsuitable adapters with a vLinker FS USB. v1.1 ATPPS [FAIL] shows, it is a very bad copy.

like this ?

OBD2 Car Diagnostic Scanner, Ferramentas de Interface, Ford FORScan, Mazda VLinker FS, USB FS, ELM327, ELM 327, OBD 2, OBDII, HS, MS-CAN
https://a.aliexpress.com/_EQYJOLf

KarelSvo, 2024-04-03:
yes