cedricp/ddt4all #1069: TPMS disable error

DavidIgor, 2024-04-05:
[image]
when I set TPMS to "without TPMS" it shows this on display and nothing is available. ironically tpms warning light goes away ...

DavidIgor, 2024-04-05:
Hi @AndyIreland it is dokker and I have done it only in cluster (as per your photo). I was looking for another option in other ecus but can't find tpms.(only in cluster)

AndyIreland, 2024-04-05:
> Hi @AndyIreland it is dokker and I have done it only in cluster (as per your photo). I was looking for another option in other ecus but can't find tpms.(only in cluster)

Yes you need to disable it in BCM/UCH ecu as well

DavidIgor, 2024-04-05:
with scan option ill get just these ecus:
[image]

matzel82, 2024-04-05:
BCM....

DavidIgor, 2024-04-05:
will check tomorrow, but I haven't seen anything tpms related in there ...  also why is it red? (I do have elm327 with can switch to read multimedia ecus ...

matzel82, 2024-04-05:
[image]

take a look here please. Is there a „with“ change it to „without“

DavidIgor, 2024-04-06:
right! all sorted you were right. thanks for help