cedricp/ddt4all #1070: Upc Renault Talisman

JohnnyX99, 2024-04-07:
Hello
I buy an used Upc for my Talisman I try from Renolink to change vin was unsuccessful and now I can't access the modul I receive an error incorrect data  string or something like that .i use the option from change vin number in Renolink.Its possible to make something now ?

JohnnyX99, 2024-04-07:
[image]
[image]
Pictura with Usm Upc

matzel82, 2024-04-07:
You have try to change the vin. This is not possible. To use Renolink (I have it too, but only not the version from the original developer 🫤) can make problems. Better you buy a new original UPC and your Garage install the latest SW and config it. So it would be work without problems.

JohnnyX99, 2024-04-07:
I have the original version of Renolink 2.05 with Obdlink Ex original