cedricp/ddt4all #1105: DDT4ALL

TotoRead, 2024-05-14:
Bonsoir ,

J'aimerai débloqué les fonctionnalités 'cachées" mirrorlink et androidauto de Rlink1 sur ma mégane 3 phase 3 estate , j'ai téléchargé DDT4ALL via ce liens , https://www.drive2.ru/b/498093336985338243/ , sauf que quand je veux choisir mon model de voiture je ne trouve pas de mégane 3 phase 3 , ça s'arrête à la phase 2.

[image]

Est-ce qu'il y aura une incidence ?

Comment mettre à jour pour trouver la phase 3 du model dans la liste ?

Merci par avance

AndyIreland, 2024-05-14:
select navigation ecu manually from the list mfd v5.4 

good luck

[image]

AndyIreland, 2024-05-14:
> Does it not bother me if I select the Mégane 3 Ph2 model when my car is a Mégane 3 ph3?

nope - as long as it definitely a rlink1 and not an rlink2 and that you have the correct rewired cable to read the rlink1 through the OBD2 socket and that the ecu can be read OK and not in red with 'NO DATA' - you should be good to go ... whatever you do, never press the 'soft reset' button in DDT4ALL on the ECU , you will have a right pain in the head getting it working again