cedricp/ddt4all #1112: How to run ddt4all on MacOS Intel

R22354413, 2024-05-20:
Hi, I am trying to open ddt4all on my macbook Pro Monterey Ver 12.6.5. I have downloaded the latest python3 and tried running the main.py via python 3 but am getting the current error

FileNotFoundError: [Errno 2] No such file or directory: 'ddt4all_data/config.json'

I have tried creating a dmg but that has failed as well. 

Does anyone have any solutions?

Am new to this so please be kind.

Furtif, 2024-05-20:
> Hi, I am trying to open ddt4all on my macbook Pro Monterey Ver 12.6.5. I have downloaded the latest python3 and tried running the main.py via python 3 but am getting the current error [...]

Yep dmg gen is issued, runs 
runs directly code:
```python3 -m pip3 install -r requirements.txt```
```python3 ./main.py```

Furtif, 2024-05-20:
Works well in mac intel i have macOs also:

[image]

R22354413, 2024-05-22:
thank you for your help. Do i run it directly on Terminal? I am getting an error saying there is no such module pip3
/usr/local/bin/python3: No module named pip3

Furtif, 2024-05-22:
> thank you for your help. Do i run it directly on Terminal? I am getting an error saying there is no such module pip3 /usr/local/bin/python3: No module named pip3

uses:
```python3 -m pip3install -r requirements.txt ```

Furtif, 2024-05-22:
or uses the sh for create a venv :

In terminal in path clone repos type:
```sh ./ddt4all.sh```

R22354413, 2024-05-26:
thank you very much . I've got it. 
[image]

Furtif, 2024-05-26:
worked ? close issue.

closed 2024-05-28 by R22354413